The-OpenROAD-Project/OpenROAD-flow-scripts #2649: Error occurred while building openroad locally

jhyun1-shim, 2025-01-03:
### Subject

[Build] for compilation-related issues.

### Describe the bug

After commit bdecdd022afa51dcc68599a80c4747ef2d1a5472,
I got the error messages while trying to build openroad locally.
Before commit, I successfully built openroad but i got a problem with the latest version.

[ 91%] Building CXX object src/dst/CMakeFiles/dst.dir/src/Worker.cc.o
In file included from /home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.h:33,
                 from /home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:29:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/WorkerConnection.h:46:36: error: expected ‘)’ before ‘&’ token
   46 |   WorkerConnection(asio::io_service& io_service,
      |                   ~                ^
      |                                    )
In file included from /home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:29:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.h:53:9: error: ‘io_service’ in namespace ‘asio’ does not name a type; did you mean ‘has_service’?
   53 |   asio::io_service service_;
      |         ^~~~~~~~~~
      |         has_service
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc: In member function ‘void dst::Worker::start_accept()’:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:43:46: error: ‘service_’ was not declared in this scope
   43 |       = boost::make_shared<WorkerConnection>(service_, dist_, logger_, this);
      |                                              ^~~~~~~~
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc: In constructor ‘dst::Worker::Worker(dst::Distributed*, utl::Logger*, const char*, short unsigned int)’:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:54:17: error: ‘service_’ was not declared in this scope
   54 |     : acceptor_(service_, tcp::endpoint(ip::address::from_string(ip), port)),
      |                 ^~~~~~~~
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:54:54: error: ‘from_string’ is not a member of ‘boost::asio::ip::address’
   54 |     : acceptor_(service_, tcp::endpoint(ip::address::from_string(ip), port)),
      |                                                      ^~~~~~~~~~~
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc: In destructor ‘dst::Worker::~Worker()’:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:63:3: error: ‘service_’ was not declared in this scope
   63 |   service_.stop();
      |   ^~~~~~~~
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc: In member function ‘void dst::Worker::run()’:
/home/jhyun1shim/GIT/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/Worker.cc:67:3: error: ‘service_’ was not declared in this scope
   67 |   service_.run();
      |   ^~~~~~~~
make[2]: *** [src/dst/CMakeFiles/dst.dir/build.make:128: src/dst/CMakeFiles/dst.dir/src/Worker.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:6518: src/dst/CMakeFiles/dst.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

### Expected Behavior

Build openroad successfully

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.167.4-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.30.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17941-g7fb347f37
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
[... 175 lines cut ...]
CMake Warning (dev) at src/dft/test/CMakeLists.txt:20 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

ARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found suitable version "1.87.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (2.5s)
-- Generating done (0.7s)
-- Build files have been written to: /tmp/tmp.2eaBv1DD85
```

### To Reproduce

sudo ./build_openroad.sh --local --threads 1

vvbandeira, 2025-02-13:
Could you try to run the following to exclude caching issues?

```
./build_openroad.sh --local --threads 1 --clean
```

Or

```
./build_openroad.sh --local --threads 1 --clean-force
```

Note that `sudo` should not be required for `build_openroad.sh` unless this is specifically required for your system.

vvbandeira, 2025-08-22:
Close due to inactivity. Please reopen if the problem is still happening on the latest version.