The-OpenROAD-Project/OpenROAD-flow-scripts #2656: ihp-sg13g2: "Horizontal congestion" violations in IO cells

dnltz, 2025-01-05:
### Subject

[Stage]: Global Router.

### Describe the bug

Global routing reports errors (GRT-0116) related to horizontally laid out IO cells. These violations are for IO to core connections. However, they're not even where the pins are located. Vertically laid out cells are fine.

### Expected Behavior

-

### Environment

```shell
I'm using a pre-built OpenROAD package: https://github.com/Precision-Innovations/OpenROAD/releases/tag/2024-12-14
and the (current) ORFS master rev (1cbce355ea39aed303f6354d8b285aec232483e0).
```

### To Reproduce

Check-out https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2655

Run `make DESIGN_CONFIG=./designs/ihp-sg13g2/i2c-gpio-expander/config.mk`. It should stop after ~ 2mins after global routing.

### Relevant log output

```shell
OpenROAD v2.0-17598-ga008522d8 
Features included (+) or not (-): +Charts +GPU +GUI +Python : None
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 16 thread(s).
global_route -congestion_report_file ./reports/ihp-sg13g2/i2c-gpio-expander/base/congestion.rpt -congestion_iterations 30 -congestion_report_iter_step 5 -verbose
[INFO GRT-0020] Min routing layer: Metal2
[INFO GRT-0021] Max routing layer: TopMetal2
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer Metal1  Track-Pitch = 0.4800  line-2-Via Pitch: 0.3650
[INFO GRT-0088] Layer Metal2  Track-Pitch = 0.4200  line-2-Via Pitch: 0.4100
[INFO GRT-0088] Layer Metal3  Track-Pitch = 0.4800  line-2-Via Pitch: 0.4100
[INFO GRT-0088] Layer Metal4  Track-Pitch = 0.4200  line-2-Via Pitch: 0.4100
[INFO GRT-0088] Layer Metal5  Track-Pitch = 3.4800  line-2-Via Pitch: 0.6200
[INFO GRT-0088] Layer TopMetal1 Track-Pitch = 2.2800  line-2-Via Pitch: 3.4100
[INFO GRT-0088] Layer TopMetal2 Track-Pitch = 4.0000  line-2-Via Pitch: 3.9500
[INFO GRT-0019] Found 50 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 17
[INFO GRT-0003] Macros: 0
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0004] Blockages: 408

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
Metal1     Vertical            0             0          0.00%
Metal2     Horizontal     359455        172851          51.91%
Metal3     Vertical       317115        151992          52.07%
Metal4     Horizontal     359455        173636          51.69%
Metal5     Vertical        42195         10268          75.67%
TopMetal1    Horizontal      42485         10215          75.96%
TopMetal2    Vertical        21315         10786          49.40%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 7246
[INFO GRT-0198] Via related Steiner nodes: 148
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 9142
[INFO GRT-0112] Final usage 3D: 33192

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
Metal1               0             0            0.00%             0 /  0 /  0
Metal2          172851          2698            1.56%             1 /  0 /  1
Metal3          151992          2873            1.89%             0 /  0 /  0
Metal4          173636            65            0.04%             0 /  0 /  0
Metal5           10268            53            0.52%             0 /  0 /  0
TopMetal1          10215            45            0.44%             1 /  0 /  5
TopMetal2          10786            32            0.30%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total           529748          5766            1.09%             2 /  0 /  6

[INFO GRT-0018] Total wirelength: 63057 um
[INFO GRT-0014] Routed nets: 1469
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
Error: global_route.tcl, 110 GRT-0116
Command exited with non-zero status 1
Elapsed time: 0:01.19[h:]min:sec. CPU time: user 1.12 sys 0.06 (99%). Peak memory: 215036KB.
```

```
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_gpio_5_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (309.6000, 316.8000) - (316.8000, 324.0000) on Layer -
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_gpio_6_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (309.6000, 468.0000) - (316.8000, 475.2000) on Layer -
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_gpio_7_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (309.6000, 547.2000) - (316.8000, 554.4000) on Layer -
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_address_0_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (727.2000, 518.4000) - (734.4000, 525.6000) on Layer -
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_address_1_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (727.2000, 583.2000) - (734.4000, 590.4000) on Layer -
violation type: Horizontal congestion
	srcs: net:sg13g2_IOPad_io_address_2_p2c
	comment: capacity:0 usage:1 overflow:1
	bbox = (727.2000, 705.6000) - (734.4000, 712.8000) on Layer -
```
```

### Screenshots

![image](https://github.com/user-attachments/assets/cdf4360a-0461-4db3-8a99-15e2217e6fd1)
![image](https://github.com/user-attachments/assets/69fea2c2-1a4b-4af8-b2c8-77188732c073)

dnltz, 2025-01-05:
[image]

dnltz, 2025-01-05:
[image]

dnltz, 2025-01-05:
Uploaded screenshots again.

dnltz, 2025-01-05:
This issue blocks https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2655

dnltz, 2025-01-05:
@maliberty sorry, cannot provide too much information since I'm using a pre-built version of OpenROAD. However, my design to reproduce this issue is in a pull-request and only takes a minute  to  build.

closed 2025-01-07 by eder-matheus

eder-matheus, 2025-01-07:
@dnltz Fixed, please checkout to the latest master branch to get the fix for this issue. Let me know if you face other problems.

dnltz, 2025-01-09:
@eder-matheus thanks a lot! I will try it