The-OpenROAD-Project/OpenROAD-flow-scripts #2714: [WARNING ODB-0126] error: polygons in special nets are not supported

dirigent-JHK, 2025-01-23:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/2713

<div type='discussions-op-text'>

<sup>Originally posted by **dirigent-JHK** January 23, 2025</sup>
Hello.
I'm trying to read LEF file, DEF file(version 5.8) via read_lef/read_def.
When trying read_def, the following error occurs;
[WARNING ODB-0126] error: polygons in special nets are not supported

My def file includes the followings;

VERSION 5.8 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
...
SPECIALNETS 1 ;
+ POLYGON LM6 ( 65421828 8188475 ) ( 65426128 8188475 ) ( 65426128 8192390 ) ( 65430564 8192390 ) ( 65430564 8188475 ) ...
...
That is,
many POLYGON's
and each POLYGON includes many points.

Is none of POLYGON supported in OpenROAD?
If so, is there any workaround for this?
</div>

dirigent-JHK, 2025-01-23:
After commenting "UNSUPPORTED("polygons in special nets are not supported");"
in definReader.cpp, the following segmentation fault occurs;

Signal 11 received
Stack trace:
 0# 0x000063AB9D263D74 in openroad
 1# 0x0000721CE4642520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBox::setLayerMask(unsigned int) in openroad
 3# odb::definSNet::rect(char const*, int, int, int, int, char const*, unsigned int) in openroad
 4# odb::definReader::specialNetCallback(DefParser::defrCallbackType_e, DefParser::defiNet*, void*) in openroad
 5# DefParser::defyyparse(DefParser::defrData*) in openroad
 6# DefParser::defrRead(_IO_FILE*, char const*, void*, int) in openroad
 7# odb::definReader::createBlock(char const*) in openroad
 8# odb::definReader::createChip(std::vector<odb::dbLib*, std::allocator<odb::dbLib*> >&, char const*, odb::dbTech*) in openroad
 9# odb::defin::createChip(std::vector<odb::dbLib*, std::allocator<odb::dbLib*> >&, char const*, odb::dbTech*) in openroad
10# ord::OpenRoad::readDef(char const*, odb::dbTech*, bool, bool, bool, bool) in openroad
11# read_def_cmd(char const*, char const*, bool, bool, bool, bool) in openroad
12# 0x000063AB9D38D66F in openroad
13# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
14# 0x0000721CE9347028 in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
17# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in openroad
18# gui::CmdInputWidget::handleEnterKeyPress(QKeyEvent*) in openroad
19# gui::TclCmdInputWidget::keyPressEvent(QKeyEvent*) in openroad
20# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
22# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
23# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# gui::SafeApplication::notify(QObject*, QEvent*) in openroad
25# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
26# 0x0000721CE63CC02F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
27# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
28# gui::SafeApplication::notify(QObject*, QEvent*) in openroad
29# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
31# 0x0000721CE00600B4 in /usr/lib/x86_64-linux-gnu/qt5/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so
32# 0x0000721CE56F1793 in /lib/x86_64-linux-gnu/libQt5Core.so.5
33# QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) in /lib/x86_64-linux-gnu/libQt5DBus.so.5
34# QObject::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
36# gui::SafeApplication::notify(QObject*, QEvent*) in openroad
37# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
38# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
39# 0x0000721CE5713A67 in /lib/x86_64-linux-gnu/libQt5Core.so.5
40# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
41# 0x0000721CE49712B8 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
42# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
43# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
44# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
45# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
46# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool, bool) in openroad
47# 0x000063AB9D265DAE in openroad
48# ord::tclInit(Tcl_Interp*) in openroad
49# ord::tclAppInit(Tcl_Interp*) in openroad
50# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
51# main in openroad
52# 0x0000721CE4629D90 in /lib/x86_64-linux-gnu/libc.so.6
53# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
54# _start in openroad
Segmentation fault (core dumped)

maliberty, 2025-01-23:
Can you provide a small test case reproducing the crash?

You should be aware of the -continue_on_errors  (though that won't help with crashing)

dirigent-JHK, 2025-01-23:
Unfortunately, I cannot provide the details of LEF, DEF due to confidential issue.
When I try another small DEF file with "-continue_on_errors",
it works, although "[WARNING ODB-0126] error: polygons in special nets are not supported" appeared.
So, I think this problem is from large data, not the polygons.
Thank you for your info on "continue_on_errors".
Because I cannot provide the details of test case, I'll clase this issue.

Thanks

closed 2025-01-23 by dirigent-JHK