The-OpenROAD-Project/OpenROAD-flow-scripts #2757: Ubuntu 22.04 local build fails

mguthaus, 2025-01-31:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Default build doesn't work. It immediately fails building yosys:
> $ ./build_openroad.sh --local
> [INFO FLW-0028] Compiling with 20 threads.
> [INFO FLW-0027] Saving logs to build_openroad.log
> [INFO FLW-0028] ./build_openroad.sh --local
> [INFO FLW-0002] Updating git submodules.
> [INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
> [INFO FLW-0017] Compiling Yosys.
> make: Entering directory '/home/mrg/OpenROAD-flow-scripts/tools/yosys'
> [  0%] Building kernel/driver.o
> [  0%] Building kernel/register.o
> [  1%] Building kernel/rtlil.o
> [  1%] Building kernel/log.o
> [  1%] Building kernel/calc.o
> In file included from kernel/driver.cc:20:
> In file included from ./kernel/yosys.h:40:
> ./kernel/yosys_common.h:23:10: fatal error: 'map' file not found
> #include <map>
>          ^~~~~

### Expected Behavior

I would expect it to compile a local version of OpenROAD.

### Environment

```shell
mrg@varactor ~/OpenROAD-flow-scripts/tools/OpenROAD ((HEAD detached at c4ec647a4))$ ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-49-generic
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-18676-gc4ec647a4
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
[... 114 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu/tclreadline.h
-- CUDD library: /usr/local/lib/libcudd.a
-- CUDD header: /usr/local/include/cudd.h
-- SSTA: 0
-- Found SWIG: /usr/local/bin/swig (found suitable version "4.1.0", minimum required is "3.0")
-- STA executable: /home/mrg/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local
```

### Relevant log output

```shell
[build_openroad.log](https://github.com/user-attachments/files/18622632/build_openroad.log)
```

maliberty, 2025-02-01:
Have you run the dependency installer?  map is provided by the compiler so it is surprising it wouldn't be found?

mguthaus, 2025-02-01:
Yes, that is called from setup.sh. I'm following the instructions in the documentation here: 

https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

drewranck, 2025-02-24:
I discovered I have this issue too. Ubuntu 22.04.5 LTS.

Outside or OpenROAD, I have been using Yosys built myself for a while w/out issue, but I also followed the instructions on Yosys's page and built it w/ gcc:

```
make config-gcc
make
```

Guess I can hack this in locally? Can I Bring-My-Own Yosys install to OpenROAD (I was surprised I had to rebuild it at all, are Yosys plugins involved?)

drewranck, 2025-02-24:
Related Yosys issue I came across before finding this issue:
https://github.com/YosysHQ/yosys/issues/2011

drewranck, 2025-02-24:
And here's a workaround that worked for me:

```
$ git diff
diff --git a/build_openroad.sh b/build_openroad.sh
index 604985e3..c7150cff 100755
--- a/build_openroad.sh
+++ b/build_openroad.sh
@@ -21,7 +21,7 @@ OPENROAD_APP_BRANCH="master"
 INSTALL_PATH="$(pwd)/tools/install"
 
 YOSYS_USER_ARGS=""
-YOSYS_ARGS="CONFIG=clang"
+YOSYS_ARGS="CONFIG=gcc"
 
 OPENROAD_APP_USER_ARGS=""
 OPENROAD_APP_ARGS=""
```

closed 2025-08-22 by vvbandeira