The-OpenROAD-Project/OpenROAD-flow-scripts #2760: Dockerhub has old image of openroad/flow-ubuntu22.04-builder

mguthaus, 2025-02-01:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Docker hub as a 9+ month old image of this project. I suggest that it be removed otherwise it confuses people. It seems like it is no longer updated. I submited a PR ( https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2759 ) to include using the image tag to prevent people from using this stale image.

### Expected Behavior

I would expect a recent image rather than an old one.

### Environment

```shell
All environments
```

### To Reproduce

Run this:

docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22.04-builder

### Relevant log output

```shell
$ docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22.04-builder
Unable to find image 'openroad/flow-ubuntu22.04-builder:latest' locally
latest: Pulling from openroad/flow-ubuntu22.04-builder
6eae748569ee: Pull complete
b304dd8a2331: Pull complete
ab0b2001f7c2: Pull complete
56f6e60563fd: Pull complete
dfdcc530d846: Pull complete
de31ef3d36f3: Pull complete
bb3310e41db1: Pull complete
4f4fb700ef54: Pull complete
c320075afe4d: Pull complete
Digest: sha256:5d4d20d3d5638fa5c79b05c6f9b240c7e5691bf8b6c47c9a46d9bb0e0d53bb09
Status: Downloaded newer image for openroad/flow-ubuntu22.04-builder:latest
groups: cannot find name for group ID 1000
I have no name!@35a672c0b7c0:/OpenROAD-flow-scripts$
exit
$ docker images | grep builder | grep latest
openroad/flow-ubuntu22.04-builder   latest    327ee7a6907a   9 months ago   16.6GB
```

mguthaus, 2025-04-15:
@vvbandeira Any suggestion to fix this? Can we tag new builds each time?

maliberty, 2025-04-16:
@sombraSoft can you help with this?