The-OpenROAD-Project/OpenROAD-flow-scripts #2763: PLACE_DENSITY_LB_ADDON is really low for asap7/swerv_wrapper

oharboe, 2025-02-02:
### Description

Taking bazel-orfs for a spin, I did a sweep of asap7/swerv_wrapper for PLACE_DENSITY_LB_ADDON, 0.20 is really low, also ROUTING_LAYER_A1DJUSTMENT = 0.2 is very low. I think this points to global routing being hard in the past.

I wonder what the purpose of the asap7/swerv_wrapper ORFS CI test is. It has some fake srams, tests CCS timing model, tinkers with RTLMP variables, constrains pins to upper/lower edge: a lot going on in terms of testing, which is good, but no clear mission for this test case leaps out.

Is it more important that it has fast build times than testing the design with higher QoR?

Although curious, I didn't plot run times against PLACE_DENSITY_LB_ADDON.

[image]

One challenge I saw is that as PLACE_DENSITY_LB_ADDON is increased, the macro placement can shuffle macros around and find terrible macro placements causing low values of PLACE_DENSITY_LB_ADDON to fail, such as for 0.63:

[image]

No congestion at 0.70, macro placement is better:

[image]

Failing global route at PLACE_DENSITY_LB_ADDON 0.74 with unsurprising location of DRCs:

[image]

oharboe, 2025-02-03:
When I decrease ROUTING_LAYER_ADJUSTMENT from default to ROUTING_LAYER_ADJUSTMENT=0.2, as in config.mk in master, there are no global routing errors all the way to PLACE_DENSITY_LB_ADDON 0.89

[image]

maliberty, 2025-02-03:
Did you go through detailed routing?  Its possible that global routing might complete but detailed routing does not.

Increasing PLACE_DENSITY_LB_ADDON is fine.  As the tool improves it is a chore to update all the tests so it happens infrequently.

"ROUTING_LAYER_ADJUSTMENT = 0.2 is very low" means what?  You think we should be reducing resources more to improve routability?

oharboe, 2025-02-03:
> Did you go through detailed routing? Its possible that global routing might complete but detailed routing does not.

I'm doing some runs with 6 iterations to see what I get in terms of DRC errors in detailed routing, but of course these runs take a long time....

[image]

> Increasing PLACE_DENSITY_LB_ADDON is fine. As the tool improves it is a chore to update all the tests so it happens infrequently.
> 
> "ROUTING_LAYER_ADJUSTMENT = 0.2 is very low" means what? You think we should be reducing resources more to improve routability?

I'm not crystal clear on what ROUTING_LAYER_ADJUSTMENT does, despite updates to the manual. I thought 0.2 would leave a lot of work to detailed routing as opposed to handling routing issues in global routing?

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/f4acf0751a0b69702a9c0a43fb36faada3a69f31/flow/scripts/variables.yaml#L36

oharboe, 2025-02-03:
Will attach/adjust issue if I have more info.

Closed for my part. As @maliberty bumping placement density and removing caveats from CI tests is an ongoing chore.

closed 2025-02-03 by oharboe

oharboe, 2025-02-03:
detailed routing plot with 5 iterations and ROUTING_LAYER_ADJUSTMENT = 0.2.

Note that I replaced 0 DRC errors with 1 to make the log plot work:

[image]