The-OpenROAD-Project/OpenROAD-flow-scripts #2766: error during installation

soujanyacccir, 2025-02-03:
using wsl in Ubuntu 20.04 
while doing  ./build_openroad.sh --local showing error 
error: could not lock config file /OpenROAD-flow-scripts/.git/modules/tools/OpenROAD/config: Permission denied fatal: could not set 'core.worktree' to '../../../../tools/OpenROAD'

rovinski, 2025-02-04:
Please use the issue template. It is there for us to get the required information to help you.
Also, your issue seems related to file permissions in git. Perhaps you cloned the repository as root or you cloned it to a place which does not have write access?

vvbandeira, 2025-02-13:
Closing due to inactivity and missing issue template. Please open a new issue and follow the template if you would like further assistance.

closed 2025-02-13 by vvbandeira