The-OpenROAD-Project/OpenROAD-flow-scripts #2771: sudo etc/DependencyInstaller.sh poisons operating system

oharboe, 2025-02-03:
### Subject

[Build] for compilation-related issues.

### Describe the bug

```
$ sudo etc/DependencyInstaller.sh
$ which -a yosys
/usr/local/bin/yosys
```

### Expected Behavior

/usr/local is now poisoned by etc/DependencyInstaller.sh, I have no robust and simple way to uninstall it. The etc/DependencyInstaller.sh should only set up files in ORFS/dependencies and use package installers(apt in Ubuntu), not mangle /usr/local.

### Environment

```shell
v3.0-2247-gce1384516
```

### To Reproduce

See above

### Relevant log output

```shell

```

povik, 2025-02-03:
Similar in spirit: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4923

povik, 2025-02-03:
Looks like the other one may have been fixed actually

oharboe, 2025-02-03:
More /usr/local poison that is messing up `./build_openroad --local --latest`:

```
$ find /usr/local/ | grep cudd
/usr/local/lib/libcudd.la
/usr/local/lib/libcudd.a
/usr/local/include/cudd.h
```

oharboe, 2025-02-03:
@povik I think the problem here is that the default behavior of DependencyInstaller.sh is to poison /usr/local if it isn't passed any arguments.

I think the solution is for DependencyInstaller.sh to fail if it is run without arugments forcing the user to make a choice as to what they want to happen. It has been so long that I had forgotten that what I wanted was `sudo ./setup.sh`. I did not want to poison my /usr/local folder, the default behavior of `sudo etc/DependencyInstaller.sh`

closed 2025-02-06 by maliberty