The-OpenROAD-Project/OpenROAD-flow-scripts #2775: local build fails

harshitbabeley, 2025-02-04:
### Subject

[Build] for compilation-related issues.

### Describe the bug

after i run setup.sh the following error occurs. it did not occur previously when i tried to install openroad 3 week sago. also i have tried to set all the permissions and everything. its correct.

### Expected Behavior

setup completed successfully

### Environment

```shell
Git commit: 357ef5e4aac85df064f31d120d691daa2b271ef2
kernel: Linux 6.5.0-21-generic
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
./etc/Env.sh: line 57: cmake: command not found
```

### To Reproduce

proper output

### Relevant log output

```shell

```

### Screenshots

[image]

[image]

[image]

[image]

maliberty, 2025-02-04:
Are you using sudo?

jeffng-or, 2025-02-04:
I'm able to repro using a70d32c05bcc06850384b7063c0b1148e679cb17:

1. git clone --recursive http://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
2. cd OpenROAD-flow-scripts/
3. sudo ./setup.sh

jeffng-or, 2025-02-04:
It looks like we used to install bison either through brew (darwin), yum (RHEL) or apt-get (ubuntu/debian). Were we unable to get the newer version of bison through those mechanisms?

vvbandeira, 2025-02-05:
> It looks like we used to install bison either through brew (darwin), yum (RHEL) or apt-get (ubuntu/debian). Were we unable to get the newer version of bison through those mechanisms?

We had issues during a `sta` update and were able to get CI to pass by using a newer version of `bison`. Some versions of the OSes you mentioned did not have it available, so we install `bison` from the source if the system version is not the one we require.

harshitbabeley, 2025-02-05:
> Are you using sudo?

Yes

vvbandeira, 2025-02-06:
@harshitbabeley Could you try with the latest version? We have merged a couple of PRs that might fix your issue.

jeffng-or, 2025-02-11:
With b9b69342aaa7db951714823f633efe7a112fe859, I'm past the bison installation failure, but am hitting:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
COIN_BZ2_LIBRARY (ADVANCED)
    linked by target "lemon" in directory /tmp/DependencyInstaller-gaCGhB/lemon-graph/lemon
    linked by target "lemon" in directory /tmp/DependencyInstaller-gaCGhB/lemon-graph/lemon
    linked by target "lp_test" in directory /tmp/DependencyInstaller-gaCGhB/lemon-graph/test
    linked by target "mip_test" in directory /tmp/DependencyInstaller-gaCGhB/lemon-graph/test
    linked by target "mip_test" in directory /tmp/DependencyInstaller-gaCGhB/lemon-graph/test

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

vvbandeira, 2025-08-22:
Close due to inactivity. We are building on Ubuntu 22.04 on our CI, so it should be fixed. Please reopen if the problem is still happening on the latest version.

closed 2025-08-22 by vvbandeira