The-OpenROAD-Project/OpenROAD-flow-scripts #2785: sudo ./setup.sh fails because it tries to write into /usr/local folder

oharboe, 2025-02-05:
### Subject

[Build] for compilation-related issues.

### Describe the bug

```
$ sudo ./setup.sh 
[deleted]
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
 /usr/bin/mkdir -p '/usr/local/share/doc/bison/examples/c/calc'
 /usr/bin/install -c -m 644 m4/bison-i18n.m4 '/usr/local/share/aclocal'
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
 /usr/bin/mkdir -p '/usr/local/share/doc/bison/examples/d/calc'
make[3]: *** [Makefile:9804: install-calcxxDATA] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:9825: install-cxxDATA] Error 1
make[3]: *** [Makefile:9846: install-dist_bistromathicDATA] Error 1
make[3]: *** [Makefile:9867: install-dist_cDATA] Error 1
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
 /usr/bin/install -c src/yacc '/usr/local/bin'
 /usr/bin/mkdir -p '/usr/local/share/doc/bison/examples/c++/calc++'
make[3]: *** [Makefile:9888: install-dist_calcDATA] Error 1
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
/usr/bin/install: cannot create regular file '/usr/local/share/aclocal/bison-i18n.m4': Permission denied
make[3]: *** [Makefile:9909: install-dist_calcdDATA] Error 1
make[3]: *** [Makefile:9783: install-aclocalDATA] Error 1
/usr/bin/install: cannot create regular file '/usr/local/bin/yacc': Permission denied
/usr/bin/mkdir: cannot create directory '/usr/local/share/doc/bison': Permission denied
make[3]: *** [Makefile:4842: install-nodist_binSCRIPTS] Error 1
make[3]: *** [Makefile:9930: install-dist_calcxxDATA] Error 1
make[3]: Leaving directory '/tmp/DependencyInstaller-qDrOqa/bison-3.8.2'
make[2]: *** [Makefile:10844: install-am] Error 2
make[2]: Leaving directory '/tmp/DependencyInstaller-qDrOqa/bison-3.8.2'
make[1]: *** [Makefile:10377: install-recursive] Error 1
make[1]: Leaving directory '/tmp/DependencyInstaller-qDrOqa/bison-3.8.2'
make: *** [Makefile:10837: install] Error 2
```

### Expected Behavior

Any dependencies that ORFS needs that can't be installed by system package installers, such as apt, should go into ./dependencies *without* using sudo access permissions.

sudo access should only be used to run the system package installers.

### Environment

```shell
v3.0-2299-g176226b6f
```

### To Reproduce

See above

### Relevant log output

```shell

```

maliberty, 2025-02-05:
Dup of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2775 ?

oharboe, 2025-02-05:
> Dup of [#2775](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2775) ?

Probably. Once #2775 is fixed, I'll retry `sudo ./setup.sh` and report any issues.

closed 2025-02-05 by oharboe

philiprbrenan, 2025-03-05:
```
Saving to: ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’

or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210. 100%[===========================================================================================>]  46.78M   193MB/s    in 0.2s    

2025-03-05 18:21:26 (193 MB/s) - ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’ saved [49050271/49050271]

./tools/OpenROAD/etc/DependencyInstaller.sh: line 954: /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
azureuser@vivado:~/OpenROAD-flow-scripts$ ll /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
-rw-r--r-- 1 root root 1 Mar  5 17:03 /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
```

I am running **sudo ./setup.sh** on **Ubuntu 24** and getting:
```
line 954: /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
```

Do you have a github action which demonstrates the correct operation of the entire installation sequence?

mguthaus, 2025-04-25:
I still have this issue on the most recent version ( 38f93c618a5821e82583352bd59da0de90222dd2). My work around is to modify the permissions and re-run:
`sudo chmod +w /home/mrg/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
`