The-OpenROAD-Project/OpenROAD-flow-scripts #2800: error during running make

harshitbabeley, 2025-02-07:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

i ran the make file after setting up the .sdc and .mk file and also added MAKE CONFIG in the Makefile. I ran make and error occured at 5_1_grt.log . the error is

[INFO GRT-0018] Total wirelength: 10393062 um
[INFO GRT-0014] Routed nets: 138935
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
Error: global_route.tcl, 110 GRT-0116
Command exited with non-zero status 1
Elapsed time: 11:04.64[h:]min:sec. CPU time: user 660.69 sys 4.13 (100%). Peak memory: 1790328KB.

### Expected Behavior

end report generation after complete flow

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.0-44-generic
os: Ubuntu 24.04.1 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-18377-gd20db5db6
-- System name: Linux
-- Compiler: GNU 13.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
[... 112 lines cut ...]
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.osHemN8lns
```

### To Reproduce

[INFO GRT-0018] Total wirelength: 10393062 um
[INFO GRT-0014] Routed nets: 138935
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
Error: global_route.tcl, 110 GRT-0116
Command exited with non-zero status 1
Elapsed time: 11:04.64[h:]min:sec. CPU time: user 660.69 sys 4.13 (100%). Peak memory: 1790328KB.

### Relevant log output

```shell

```

maliberty, 2025-02-07:
Examine the congestion heat map in the GUI and paste an image here for help

harshitbabeley, 2025-02-07:
> Examine the congestion heat map in the GUI and paste an image here for help

Where can I get the heat map. Pls guide

maliberty, 2025-02-07:
https://openroad-flow-scripts.readthedocs.io/en/latest/tutorials/FlowTutorial.html#using-heat-maps

harshitbabeley, 2025-02-07:
> https://openroad-flow-scripts.readthedocs.io/en/latest/tutorials/FlowTutorial.html#using-heat-maps

but as the .def file is not generated how can i open gui.

maliberty, 2025-02-07:
The GUI uses the .odb file.

harshitbabeley, 2025-02-07:
[image]

I have opened the 5_1_grt.odb on openroad gui

maliberty, 2025-02-07:
It looks like the placement density is too high - what do you have it set to?

harshitbabeley, 2025-02-07:
Initially place-density variable was 0.6. then I reduced to 0.3 which resulted in floor plan error where it said that density to low and suggested to increase it to 0.41. I did the same and ran the flow again. The same error occurred. The above heat map is for 0.41. 
Currently I have set the die_area as (0 0 3000 3000) and place density as 0.61(lower than that was not working). It's running. I don't whether it works or not. But I don't want to do this hit and trial. It takes hours to run. 😭😭

harshitbabeley, 2025-02-07:
Can you suggest me the perfect comfig.mk file. I really need to complete this project in time. Thank you in advance

maliberty, 2025-02-08:
With no test case it is hard to help.   In order to lower the density further you may need to expand the area for the block.

harshitbabeley, 2025-02-08:
How can I do that. And what do you mean by test case

maliberty, 2025-02-08:
A test case is something that runs and reproduces the issue rather than just some text and images.

What does "I really need to complete this project in time." refer to?  Is this a school project of some sort?

You said "Currently I have set the die_area as (0 0 3000 3000)" so you would increase that area.

harshitbabeley, 2025-02-08:
Yes it's a college mini project

harshitbabeley, 2025-02-08:
How can I get and give test case to you. I am new to openROAD so I don't know much

harshitbabeley, 2025-02-08:
I don't want you to do the project. I just need a way to resolve this error. It's takes a long time run one script. And I don't think changing die area or place density manually is a good idea

maliberty, 2025-02-08:
You have a congestion problem so it will require some change.

harshitbabeley, 2025-02-08:
I am trying to do changes. But if there is a proper path to get through this error then it would be very helpful

maliberty, 2025-02-08:
Reducing the placement density is the most likely path to success.

harshitbabeley, 2025-02-08:
Based on the heat map can you recommend how much density should I have. Like some value around which I can test

harshitbabeley, 2025-02-09:
i tried to decrease density and set the MAX_ROUTE layer as met5. A new error came at gloubal_route
Running detail_route.tcl, stage 5_2_route
[INFO DRT-0149] Reading tech and libs.
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via2
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via2
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via3
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via3
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via4
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via4
Error: detail_route.tcl, 47 map::at
Command exited with non-zero status 1
Elapsed time: 0:04.85[h:]min:sec. CPU time: user 4.42 sys 0.45 (100%). Peak memory: 926008KB.
make[1]: *** [Makefile:761: do-5_2_route] Error

maliberty, 2025-02-09:
That sounds more like a bug.  How can I reproduce it?

Lowering MAX_ROUTING_LAYER will only produce more congestion as you remove possible layers to route on.

You should be looking to get the routing congestion below 100%.