The-OpenROAD-Project/OpenROAD-flow-scripts #2888: setuptools version requirement breaking Docker-based AutoTuner builds

jeffng-or, 2025-02-28:
### Subject

[Build] for compilation-related issues.

### Describe the bug

An update to the version requirement for Python3 setuptools is breaking Docker-based builds:

```
      /tmp/pip-build-env-1trg79qq/overlay/local/lib/python3.10/dist-packages/setuptools_scm/_integration/setuptools.py:31: RuntimeWarning:
      ERROR: setuptools==59.6.0 is used in combination with setuptools-scm>=8.x
      
      Your build configuration is incomplete and previously worked by accident!
      setuptools-scm requires setuptools>=61
```

I've tried a couple of workarounds to uninstall either python3-setuptools or python3-pip and nothing is working:

1. ```pip3 install --upgrade setuptools==61.0.0``` installs setuptools 61.0.0 in /usr/local/lib/python3.10/dist-packages, but leaves version 59.6.0 in /usr/lib/python3/dist-packages/setuptools. The 59.6.0 version is then used during AT requirements installation and fails
2. Uninstalling setuptools and then re-installing pip3 ends up reinstalling setuptools 59.6.0, so back to same situation as #1.

@vvbandeira / @luarss , any ideas?

### Expected Behavior

Functioning docker build like before

### Environment

```shell
commit ee66b1911acd536828c34c490e6cf62193dada3b
```

### To Reproduce

Here's the Dockerfile that was working previously

```
# syntax=docker/dockerfile:1
#
# Installs ORFS from docker image 
#

FROM openroad/orfs:v3.0-2380-g20d2f7bf

# install AT required packages
RUN pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt

# ORFS installation dir
WORKDIR /OpenROAD-flow-scripts/tools/AutoTuner/src
```

To reproduce:
```docker build -t autotuner:1.0 -f Dockerfile .```

### Relevant log output

```shell
#0 building with "default" instance using docker driver

#1 [internal] load build definition from Dockerfile.at
#1 transferring dockerfile: 338B done
#1 DONE 0.0s

#2 resolve image config for docker-image://docker.io/docker/dockerfile:1
#2 DONE 0.2s

#1 [internal] load build definition from Dockerfile.at
#1 transferring dockerfile: 338B done
#1 DONE 0.0s
[... 162 lines cut ...]
#8 9.706 note: This error originates from a subprocess, and is likely not a problem with pip.
#8 ERROR: process "/bin/sh -c pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt" did not complete successfully: exit code: 1
------
 > [2/3] RUN pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt:
9.704       [end of output]
9.704   
9.704   note: This error originates from a subprocess, and is likely not a problem with pip.
9.706 error: subprocess-exited-with-error
9.706 
9.706 × Getting requirements to build wheel did not run successfully.
9.706 ? exit code: 1
9.706 ??> See above for output.
9.706 
9.706 note: This error originates from a subprocess, and is likely not a problem with pip.
------
Dockerfile.at:9
--------------------
   7 |     
   8 |     # install AT required packages
   9 | >>> RUN pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt
  10 |     
  11 |     # ORFS installation dir
--------------------
ERROR: failed to solve: process "/bin/sh -c pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt" did not complete successfully: exit code: 1
```

vvbandeira, 2025-03-01:
Upgrading `pip` before installing the requirements allows the image building to finish.
You can use:

```
RUN pip3 install --upgrade pip
RUN pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt
```

OR

```
RUN pip3 install --upgrade pip \
      && pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt
```

However I see some error messages that are worth looking into:

> ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
> grpcio-status 1.70.0 requires protobuf<6.0dev,>=5.26.1, but you have protobuf 3.20.3 which is incompatible.

jeffng-or, 2025-03-03:
@vvbandeira , thanks! I was able to work around the issue.

Do you want me to leave the issue open for Jack to look at the other error messages you mention?

luarss, 2025-03-08:
Hi, this should be resolved by https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2859

PR #3015 (2025-03-31): [Autotuner] Bump versions 1
Part 1 to bump versions
- urllib3 for security update
- protobuf dependency issues, fixes #2888, resolves https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2832
closed 2025-03-31 by vvbandeira