The-OpenROAD-Project/OpenROAD-flow-scripts #2893: Fail to install local on fedora 41 : no rule to make target libboost_serialization.a

lionelsaintecluque, 2025-03-04:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Hello, 

I'm building local version of OpenRoad-flow-scripts on a linux Fedora 41, as local binaries. 

I did some things already : 
Install lemon from sources, 
Install OR-Tool (binaries)
Install CUDD from sources, 

Now I have an issue for which I don't find solution : 

[ 11%] Built target dst_messages
[ 11%] Built target dst
gmake[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib64/libboost_serialization.a », nécessaire pour « src/dst/test/cpp/TestDistributed ». Arrêt.
gmake[1]: *** [CMakeFiles/Makefile2:7040: src/dst/test/cpp/CMakeFiles/TestDistributed.dir/all] Error 2
gmake[1]: *** Attente des tâches non terminées....
gmake[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib64/libboost_serialization.a », nécessaire pour « src/dst/test/cpp/TestBalancer ». Arrêt.
gmake[1]: *** [CMakeFiles/Makefile2:7011: src/dst/test/cpp/CMakeFiles/TestBalancer.dir/all] Error 2
gmake[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib64/libboost_serialization.a », nécessaire pour « src/dst/test/cpp/TestWorker ». Arrêt.
gmake[1]: *** [CMakeFiles/Makefile2:6982: src/dst/test/cpp/CMakeFiles/TestWorker.dir/all] Error 2
[ 17%] Built target db
[ 65%] Built target libabc
gmake: *** [Makefile:146: all] Error 2

I'm sorry, my computer speaks french. It says "No rule to make target ..." 

Thanks for help

### Expected Behavior

.

### Environment

```shell
.
```

### To Reproduce

.

### Relevant log output

```shell

```

vvbandeira, 2025-03-04:
We don't officially support Fedora at the moment. The error message you have is due to a missing boost; there will be many other similar error messages if you do not install all the dependencies before trying to compile. I suggest looking through the following scripts and replicating the commands on your system. 

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/etc/DependencyInstaller.sh
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh

lionelsaintecluque, 2025-03-04:
Hello @vvbandeira 

Thanks for the answer. I'm not sure it is a matter of installing dependencies : boost, boost-devel and boost-serialization are installed. 

And when I list the content of my library folder I get : 
```
 ls -lha /usr/lib64/libboost_serialization*
lrwxrwxrwx. 1 root root   32 17 juil.  2024 /usr/lib64/libboost_serialization.so -> libboost_serialization.so.1.83.0
-rwxr-xr-x. 1 root root 264K 17 juil.  2024 /usr/lib64/libboost_serialization.so.1.83.0
```

Somehow gmake is looking for a **.a** library, and not a **.so**. I guess it's relevant for this problem, but I don't find proper solution to that.

vvbandeira, 2025-03-04:
Did you install `boost` from your package manager or build from source? We build from source with:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh#L146-L161

It might be the case where your package manager only provides `.so` or that you did not compiled `boost` same as we do.

An alternative is to enable System Boost:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/CMakeLists.txt#L63-L64

lionelsaintecluque, 2025-03-04:
Indeed ! Boost was installed with the package manager (dnf) 

I choosed your altrenative solution. I updated CMakeLists.txt first, but it had no effect (perhaps it's normal, I don't know how CMake works).

Then, I edited build_openroad.sh to force this option from the command line  :  
`OPENROAD_APP_ARGS+=" **-DUSE_SYSTEM_BOOST=ON** -D CMAKE_INSTALL_PREFIX=${INSTALL_PATH}/OpenROAD"`

It compiled successfully

Thanks a lot.

closed 2025-03-04 by vvbandeira