The-OpenROAD-Project/OpenROAD-flow-scripts #2904: Line 954: Permission denied

philiprbrenan, 2025-03-05:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I am running sudo ./setup.sh on Ubuntu 24 and getting:
```
line 954: /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
```
The context is:

```
Saving to: ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’

or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210. 100%[===========================================================================================>]  46.78M   193MB/s    in 0.2s    

2025-03-05 18:21:26 (193 MB/s) - ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’ saved [49050271/49050271]

./tools/OpenROAD/etc/DependencyInstaller.sh: line 954: /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
azureuser@vivado:~/OpenROAD-flow-scripts$ ll /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
-rw-r--r-- 1 root root 1 Mar  5 17:03 /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
```

Do you have a github action which demonstrates the correct operation of the entire installation sequence?

### Expected Behavior

The installation should complete without such errors

### Environment

```shell
azureuser@vivado:~/OpenROAD-flow-scripts$ ll etc*
total 36
drwxrwxr-x  2 azureuser azureuser  4096 Mar  5 17:01 ./
drwxrwxr-x 13 azureuser azureuser  4096 Mar  5 17:04 ../
-rwxrwxr-x  1 azureuser azureuser 12620 Mar  5 17:01 DependencyInstaller.sh*
-rwxrwxr-x  1 azureuser azureuser  6440 Mar  5 17:01 DockerHelper.sh*
-rwxrwxr-x  1 azureuser azureuser   786 Mar  5 17:01 DockerTag.sh*
azureuser@vivado:~/OpenROAD-flow-scripts$ bash ./etc/Env.sh
bash: ./etc/Env.sh: No such file or directory
```

### To Reproduce

```
sudo ./setup.sh
```

### Relevant log output

```shell
Saving to: ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’

or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210. 100%[===========================================================================================>]  46.78M   193MB/s    in 0.2s    

2025-03-05 18:21:26 (193 MB/s) - ‘or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz’ saved [49050271/49050271]

./tools/OpenROAD/etc/DependencyInstaller.sh: line 954: /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt: Permission denied
azureuser@vivado:~/OpenROAD-flow-scripts$ ll /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
-rw-r--r-- 1 root root 1 Mar  5 17:03 /home/azureuser/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt
```

maliberty, 2025-03-05:
Why don't you have write permission to a file in your local area?

vvbandeira, 2025-03-05:
@maliberty The issue is that the user's `openroad_deps_prefixes.txt` file is owned by `root`, so he cannot modify the file.

@philiprbrenan did you run anything other than `sudo ./setup.sh` on your workspace? When using the `setup.sh` script the files created should have ownership of the user that called the script. Are you using the latest version of ORFS?

philiprbrenan, 2025-03-06:
I deleted the folder containing ORFS, did git clone ... again to recreate the folder, ran set up, received the same error, proceeded anyway and was able to run the gcd sample successfully and locally, which was impressive. This is on Ubuntu 24.

vvbandeira, 2025-03-06:
Great to hear. I will keep the issue open for the time being. I will try to reproduce the error on a Ubuntu 24 machine; on my Ubuntu 22, I was not able.

migcorre, 2025-03-22:
'openroad_deps_prefixes.txt' is as ROOT. I solved coping this file, deleted the original, and copy the file again. This leave the file with USER instead of ROOT. 
I see that is trying to install or-tools_amd64_ubuntu-22.04_cpp_v9.11.4210.tar.gz (ubuntu 22.04 instead 24.04). to solve this goes to DependencyInstaller.sh file, line 851 and force the tool to 24.04, example:
```
_installUbuntuPackages "24.04"
_installOrTools "ubuntu" "24.04" "amd64"
```
and run ./etc/Build.sh with sudo

vvbandeira, 2025-03-24:
Why run `./etc/Build.sh` as root?

mguthaus, 2025-04-25:
I still have this issue on the most recent version ( [38f93c6](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/38f93c618a5821e82583352bd59da0de90222dd2)). My work around is to modify the permissions and re-run:
sudo chmod +w /home/mrg/OpenROAD-flow-scripts/tools/OpenROAD/etc/openroad_deps_prefixes.txt

mguthaus, 2025-04-25:
Actually, chmod and chown don't work because the file gets re-written and permissions changed. However, build_openroad.sh works fine since this seems to just document the deps and isn't really needed.

closed 2025-08-28 by vvbandeira