The-OpenROAD-Project/OpenROAD-flow-scripts #2906: ERROR: Missing cost information on instanced blackbox ram_2048x39

oharboe, 2025-03-05:
### Subject

[Stage]: Synthesis.

### Describe the bug

untar and run https://drive.google.com/file/d/1WaywLtJc3bCI4qxki1PdbsxYTOPxkhXK/view?usp=sharing

```
$ ./run-me-lsu_dccm_mem_netlist_synth-asap7-netlist.sh
[deleted]
25. Executing KEEP_HIERARCHY pass.
<suppressed ~829 debug messages>
ERROR: Missing cost information on instanced blackbox ram_2048x39
```

### Expected Behavior

No error

### Environment

```shell
ORFS v3.0-2528-ga8780221e
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-03-05:
@povik I don't know if this is an interesting problem, but I tought I might provide a test-case. It happens when I incorrectly set SYNTH_HIERARCHICAL=1 when I should be flattening in parallel synthesis.

maliberty, 2025-03-05:
see https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/2889

povik, 2025-03-05:
I'm on the edge as to whether this should be a warning or an error. The message means there's a blackbox with no area information to use for the ungrouping decisions in SYNTH_HIERARCHICAL mode

The message could be improved at least by telling people what to do to fix it. Let me assign myself to do that.

oharboe, 2025-06-17:
@povik Is this still relevant?