The-OpenROAD-Project/OpenROAD-flow-scripts #2912: Git Quickstart guide link broken

WickedStereo, 2025-03-06:
### Description

The link to the Git Quickstart guide in the file `docs/contrib/DeveloperGuide.md` is broken and returns a 404 Not Found error.

## Steps to Reproduce
1. Navigate to `docs/contrib/DeveloperGuide.md` in the repository.
2. Click on the link to the Git Quickstart guide (last point in the file)

[image]

### Suggested Solution

Update the link to a valid Git Quickstart guide.

luarss, 2025-03-07:
Hi, please refer to this link for the timebeing https://openroad-flow-scripts.readthedocs.io/en/latest/contrib/GitGuide.html

The actual file does not exist in the repo because we share the same Git Quickstart guide with OR repo.

WickedStereo, 2025-03-08:
Thanks for letting me know. Do you want me to add the doc in the ORFS repo and submit a pull request?

WickedStereo, 2025-03-08:
That's right! A simple solution is to use the absolute URL to the file from OR. The only issue is that it would always point to github and not work locally. 
I checked other possible solutions which include creating submodules or automation using github actions.

luarss, 2025-03-08:
Thats not the only thing that needs to change. 

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/conf.py

In our setup we download the markdown, do regex swap for OR into ORFS then render it.

PR #3026 (2025-04-01): [Docs] Add permalink to git guide
Fixes #2912 by adding a permalink to ORFS readthedocs site's page. Actual file should be overwritten during sphinx compilation.

See verification: https://openroad-flow-scripts--3026.org.readthedocs.build/en/3026/contrib/GitGuide.html