The-OpenROAD-Project/OpenROAD-flow-scripts #2913: OpenROAD binary release fails with the flow scripts `make` command

nachiket, 2025-03-07:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I installed the latest Dec 2024 OpenROAD binary from here: https://github.com/Precision-Innovations/OpenROAD for Ubuntu 22.04

When I run the `make` command in the OpenROAD-flow-scripts, most of the flow works except a weird STA error

```
==========================================================================
global route report_design_area
--------------------------------------------------------------------------
Design area 899 u^2 83% utilization.
[INFO FLW-0007] clock core_clock period 0.460000
[INFO FLW-0008] Clock core_clock period 0.496
[INFO FLW-0009] Clock core_clock slack -0.063
Error: write_ref_sdc.tcl, 40 invalid command name "sta::endpoint_path_count"
Command exited with non-zero status 1
Elapsed time: 0:13.40[h:]min:sec. CPU time: user 29.94 sys 0.17 (224%). Peak memory: 363672KB.
make[1]: *** [Makefile:735: do-5_1_grt] Error 1
make: *** [Makefile:733: results/nangate45/gcd/base/5_1_grt.odb] Error 2
```

### Expected Behavior

I expect the off-the-shelf `make` command to run to completion without errors. Maybe I have to revert to a Dec 2024 version of the OpenROAD-flow-scriptsr repo to align with the binary release?

### Environment

```shell
I'm using the binary release, so not sure I know what to do here.
```

### To Reproduce

Install OpenROAD binary release from Dec 2024 + latest OpenROAD-flow-srcripts + `make` command.

### Relevant log output

```shell

```

Karthikganiga443, 2025-03-09:
If we comment out line 40 in write_ref_sdc.tcl, we will encounter another error.

[image]

maliberty, 2025-03-09:
Please package and upload a test case for the crash if you can share one.   `make detail_route_issue` will create a tarball

Karthikganiga443, 2025-03-09:
I've packaged the test case using make detail_route_issue. 

[detail_route_gcd_nangate45_base_2025-03-09_13-39.tar.gz](https://github.com/user-attachments/files/19148310/detail_route_gcd_nangate45_base_2025-03-09_13-39.tar.gz)

Let me know if you need anything else!

osamahammad21, 2025-03-09:
@maliberty The DRT issue was fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/6491/files

oharboe, 2025-04-06:
@osamahammad21 I saw this error a while back, I think. Close this issue now?

maliberty, 2025-04-06:
Reopen if the problem persists

closed 2025-04-06 by maliberty