The-OpenROAD-Project/OpenROAD-flow-scripts #2924: Error: write_ref_sdc.tcl, 40 invalid command name "sta::endpoint_path_count". this is the error iam facing.

faisaltashildar, 2025-03-11:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

[ORFS_installation_ubuntu22.04.docx\[1\].pdf](https://github.com/user-attachments/files/19183424/ORFS_installation_ubuntu22.04.docx.1.pdf)

I a ECE student studying in college while following this instructions i am facing this error and. I think running a "make" final command in OpenROAD-flow-scripts/flow directory, the GUI window will open. Please help me.

### Expected Behavior

==========================================================================
global route report_design_area
--------------------------------------------------------------------------
Design area 868 u^2 80% utilization.
[INFO FLW-0007] clock core_clock period 0.460000
[INFO FLW-0008] Clock core_clock period 0.502
[INFO FLW-0009] Clock core_clock slack -0.069
Error: write_ref_sdc.tcl, 40 invalid command name "sta::endpoint_path_count"
Command exited with non-zero status 1
Elapsed time: 0:22.92[h:]min:sec. CPU time: user 27.88 sys 2.84 (134%). Peak memory: 366264KB.
make[1]: *** [Makefile:735: do-5_1_grt] Error 1
make: *** [Makefile:733: results/nangate45/gcd/base/5_1_grt.odb] Error 2

### Environment

```shell
s
```

### To Reproduce

s

### Relevant log output

```shell

```

### Screenshots

[image]

vijayank88, 2025-03-11:
@vvbandeira 
This is part of IITG-VTU PD training using OpenROAD.
Everyone using pre-built binary method.
With latest ORFS, facing above error.
I believe updating latest OpenROAD in https://github.com/Precision-Innovations/OpenROAD/releases resolve this issue.
Do the needful.
Thanks...

maliberty, 2025-03-11:
We are in the process of moving to a different method of distributing pre-built binaries.  Once that is complete we will have more regular releases.

faisaltashildar, 2025-03-11:
Can you just please send me a video of showing how to install open road and run the software correctly in a sequentially manner please it will be helpful for me

faisaltashildar, 2025-03-12:
@maliberty sir if you can assign a person to resolve this issue to complete my installation i would be very thankful sir. Thank you.

sombraSoft, 2025-03-12:
Hey @faisaltashildar, we're starting to provide public releases for ubuntu and debian at https://vaultlink.precisioninno.com/
Only thing it will ask of you is to provide your name and e-mail. Reach out if you have any questions!

faisaltashildar, 2025-03-12:
@sombraSoft what will be the process of the installation of this software sir ?? I need a oversll guide to install it in my ubuntu sir. Help in in any way if you can sir i would be very thankful sir. Thankyou

vijayank88, 2025-03-13:
@faisaltashildar let me check the URL and share you the steps.

faisaltashildar, 2025-03-13:
[image]
@vijayank88 sir there is no OpenROAD-flow-scripts folder in my ubuntu system. i followed the same steps which you told and installed the newest version once again.

faisaltashildar, 2025-03-13:
@vijayank88 @sombraSoft @maliberty Thank you for your time and support sir. I have successfully installed the OpenROAD software in my PC.

[image]
[image]

closed 2025-03-13 by vvbandeira