The-OpenROAD-Project/OpenROAD-flow-scripts #2932: Unable to update OpenROAD pre-built binary for Ubuntu20.04

vijayank88, 2025-03-13:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I have downloaded Ubuntu20.04 OpenROAD binary from here: https://vaultlink.precisioninno.com/

If I try to install, it fails with below error:
```
sudo apt install ./openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb 
[sudo] password for vijayan: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'openroad' instead of './openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb'
The following packages will be upgraded:
  openroad
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/54.6 MB of archives.
After this operation, 7,417 kB of additional disk space will be used.
Get:1 /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb openroad amd64 2.0-19576-gec1bf1a13 [54.6 MB]
(Reading database ... 445195 files and directories currently installed.)
Preparing to unpack .../openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb ...
Unpacking openroad (2.0-19576-gec1bf1a13) over (2.0-17598-ga008522d8) ...
dpkg: error processing archive /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb (--unpack):
 trying to overwrite '/usr/include/gtest/gtest-death-test.h', which is also in package libgtest-dev:amd64 1.10.0-2
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### Expected Behavior

Install OpenROAD successfully using pre-built binaries

### Environment

```shell
Using Prebuilt binaries from PII vault.
```

### To Reproduce

Update OpenROAD binary with Ubuntu 20.04 OS.
```
sudo apt install ./openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb
```

### Relevant log output

```shell
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'openroad' instead of './openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb'
The following packages will be upgraded:
  openroad
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/54.6 MB of archives.
After this operation, 7,417 kB of additional disk space will be used.
Get:1 /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb openroad amd64 2.0-19576-gec1bf1a13 [54.6 MB]
(Reading database ... 445195 files and directories currently installed.)
Preparing to unpack .../openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb ...
Unpacking openroad (2.0-19576-gec1bf1a13) over (2.0-17598-ga008522d8) ...
dpkg: error processing archive /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb (--unpack):
 trying to overwrite '/usr/include/gtest/gtest-death-test.h', which is also in package libgtest-dev:amd64 1.10.0-2
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /home/vijayan/PII/OR_Packages/openroad_2.0-19576-gec1bf1a13_amd64-ubuntu-20.04.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

vvbandeira, 2025-03-17:
@vijayank88 
Can you remove the conflicting package? `libgtest-dev`
We will try to remove the dependency in future releases, as this is only for testing.

vijayank88, 2025-03-20:
@vvbandeira 
Thanks for the reply.
By using `sudo apt-get purge libgtest-dev` removed the conflict package and able to install latest OpenROAD successfully.

closed 2025-03-20 by vvbandeira