The-OpenROAD-Project/OpenROAD-flow-scripts #2935: Re-tighten hold violations count for ngt45/bp_fe_top

AcKoucher, 2025-03-13:
### Description

Re-tighten the temporary metrics update in #2934.

### Suggested Solution

This should be viable once [#6846](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6846) is merged.

More details in [here](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6834#issuecomment-2718770527) and [here](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6834#issuecomment-2722305685).

arthurjolo, 2025-03-24:
PR [#6846](https://github.com/The-OpenROAD-Project/OpenROAD/pull/6846) was merged, but the metrics had already been re-tighten.

AcKoucher, 2025-03-24:
I still see the high limit on master:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/7494151741c8baf573bd3ddb0f681d6c600b6fb0/flow/designs/nangate45/bp_fe_top/rules-base.json#L58-L61

AcKoucher, 2025-03-24:
Apparently it has to do with the OR version update. I'll keep this issue opened until the limit is back to a lower value.