The-OpenROAD-Project/OpenROAD-flow-scripts #2937: Issue 'map' file not found

farhan-sw, 2025-03-14:
I have successfully installed the dependencies in the script. However, when building, I encountered the error 'map' file not found.

> ./kernel/yosys_common.h:23:10: fatal error: 'map' file not found

[build_openroad.log](https://github.com/user-attachments/files/19243225/build_openroad.log)

vvbandeira, 2025-03-14:
Please follow the issue template so we can better help you.
What is your OS? What commands did you run to install the dependencies? What is the version of ORFS you are using?

mguthaus, 2025-04-17:
This is related to yosys issue:
https://github.com/YosysHQ/yosys/issues/4244

maliberty, 2025-04-17:
If you wish you can PR the dependency installer as it sounds like you have a machine to test on that shows the issue.

vvbandeira, 2025-08-22:
Support for Ubuntu 24.04 has been improved in the latest master. Please reopen if the problem is still happening.

closed 2025-08-22 by vvbandeira