The-OpenROAD-Project/OpenROAD-flow-scripts #2941: Synthesis fails because it doesn't strip $display("sdram: unmapped addr=%x",XADDR); statements

oharboe, 2025-03-15:
### Subject

[Stage]: Synthesis.

### Describe the bug

untar and run https://drive.google.com/file/d/1aqUP71geuLoM0J608i5RZ9ytAoZOteDY/view?usp=sharing

```
Checking module darksocv...
Warning: Cell darksocv.$display$rtl/darksocv.v:294$4358 is an unmapped internal cell of type $print.
Checking module darkuart...
Found and reported 1 problems.
ERROR: Found 1 problems in 'check -assert'.
```

### Expected Behavior

$display() statements should be stripped before `check -assert` check:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/88a6db4d85587b48b34fa37029194cf5c5564271/flow/scripts/synth.tcl#L125-L133

### Environment

```shell
v3.0-2590-g22101c1af
```

### To Reproduce

See above

### Relevant log output

```shell

```

oharboe, 2025-03-15:
@povik Is there a command missing to strip $displa() and similar or ignore it in the check?

povik, 2025-03-15:
@oharboe yes, we should add `delete t:\$print` to this part of synth.tcl
```
# Get rid of indigestibles
chformal -remove
```
Related: yosyshq/yosys#4122

PR #2942 (2025-03-15): synth: clean out more indigestibles
fixes #2941
commit 89bf4ce6 (2025-03-15): synth: clean out more indigestibles

fixes #2941
closed 2025-03-15 by maliberty