The-OpenROAD-Project/OpenROAD-flow-scripts #3034: Flow finishing fails without GUI

UDXS, 2025-04-02:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

When OpenROAD is built with GUI, the flow expects GUI backend to be available. This is not always the case on a lot of university/lab-type shared UNIX instances that rely on optional X forwarding.

### Expected Behavior

Either OR or ORFS should fail out gracefully if GUI is unavailable and skip the generation of images.

### Environment

```shell
N/A
```

### To Reproduce

[save_images_gcd_ihp-sg13g2_base_2025-04-02_15-52.tar.gz](https://github.com/user-attachments/files/19576876/save_images_gcd_ihp-sg13g2_base_2025-04-02_15-52.tar.gz)

### Relevant log output

```shell
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 23796 u^2 98% utilization.
[WARNING GUI-0076] could not connect to display docker.for.mac.host.internal:0
[INFO GUI-0075] Could not load the Qt platform plugin "xcb" in "" even though it was found.
[ERROR GUI-0077] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

terminate called after throwing an instance of 'std::runtime_error'
  what():  GUI-0077
Signal 6 received
Stack trace:
 0# 0x000055555635FF0D in /usr/bin/openroad
 1# 0x00007FFFFB0F7520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007FFFFB489B9E in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x00007FFFFB49520C in /lib/x86_64-linux-gnu/libstdc++.so.6
 7# 0x00007FFFFB4941E9 in /lib/x86_64-linux-gnu/libstdc++.so.6
 8# __gxx_personality_v0 in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# 0x00007FFFFB2F4884 in /lib/x86_64-linux-gnu/libgcc_s.so.1
10# _Unwind_Resume in /lib/x86_64-linux-gnu/libgcc_s.so.1
11# 0x00007FFFFBC7A2F9 in /lib/x86_64-linux-gnu/libQt5Core.so.5
12# QMessageLogger::fatal(char const*, ...) const in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# QGuiApplicationPrivate::createPlatformIntegration() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
14# QGuiApplicationPrivate::createEventDispatcher() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
15# QCoreApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Core.so.5
16# QGuiApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
17# QApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool, bool) in /usr/bin/openroad
19# gui::Gui::showGui(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) in /usr/bin/openroad
20# show(char const*, bool, bool) in /usr/bin/openroad
21# 0x0000555556A7367B in /usr/bin/openroad
22# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
23# 0x00007FFFFF649028 in /lib/x86_64-linux-gnu/libtcl8.6.so
24# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
25# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
26# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /usr/bin/openroad
27# 0x0000555556360AD4 in /usr/bin/openroad
28# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
29# main in /usr/bin/openroad
30# 0x00007FFFFB0DED90 in /lib/x86_64-linux-gnu/libc.so.6
31# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
32# _start in /usr/bin/openroad
Command terminated by signal 6
Elapsed time: 0:02.13[h:]min:sec. CPU time: user 1.98 sys 0.12 (98%). Peak memory: 182696KB.
make[1]: *** [Makefile:800: do-6_report] Error 134
make: *** [Makefile:798: logs/ihp-sg13g2/gcd/base/6_report.log] Error 2
```

vvbandeira, 2025-07-09:
@UDXS, when running OR without a proper X setup, you can `export QT_QPA_PLATFORM=offscreen`, this allows some GUI functionality to work without X

harsh-kumar-patwa, 2026-03-02:
Raised a PR for this — #3944

Added a `DISPLAY` environment check in `final_report.tcl` so `save_images` is skipped on headless machines instead of crashing. Uses the existing `env_var_exists_and_non_empty` helper already used elsewhere in the same file.

Let me know if anything needs adjusting!