The-OpenROAD-Project/OpenROAD-flow-scripts #3041: ERROR: TCL interpreter returned an error: Yosys command produced an error

mithro, 2025-04-03:
### Subject

[Stage]: Synthesis.

### Describe the bug

I'm trying to follow the "OpenROAD Flow Scripts Tutorial" instructions found at https://openroad-flow-scripts.readthedocs.io/en/latest/tutorials/FlowTutorial.html using the https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html system.

At the [Verifying the Docker based Installation](https://openroad-flow-scripts.readthedocs.io/en/latest/tutorials/FlowTutorial.html#verifying-the-docker-based-installation) step I'm getting the following output;
```
# ==============================================================================
#  ____  _____ _____ _   _ ____
# / ___|| ____|_   _| | | |  _ \
# \___ \|  _|   | | | | | | |_) |
I have no name!@ebbdfe7a8b0a:/OpenROAD-flow-scripts/flow$ make
/OpenROAD-flow-scripts/flow/scripts/synth.sh /OpenROAD-flow-scripts/flow/scripts/synth_canonicalize.tcl ./logs/nangate45/gcd/base/1_1_yosys_canonicalize.log
1. Executing Verilog-2005 frontend: /OpenROAD-flow-scripts/flow/designs/src/gcd/gcd.v
2. Executing Liberty frontend: ./objects/nangate45/gcd/base/lib/NangateOpenCellLibrary_typical.lib
ERROR: TCL interpreter returned an error: Yosys command produced an error
Command exited with non-zero status 1
Elapsed time: 0:00.15[h:]min:sec. CPU time: user 0.14 sys 0.01 (100%). Peak memory: 32960KB.
make[1]: *** [Makefile:291: do-yosys-canonicalize] Error 1
make: *** [Makefile:294: results/nangate45/gcd/base/1_synth.rtlil] Error 2
I have no name!@ebbdfe7a8b0a:/OpenROAD-flow-scripts/flow$ 
```

### Expected Behavior

Get the following output as described in the document;
```
[INFO] Writing out GDS/OAS 'results/nangate45/gcd/base/6_1_merged.gds'
Elapsed time: 0:10.44[h:]min:sec. CPU time: user 2.17 sys 0.54 (26%). Peak memory: 274184KB.
cp results/nangate45/gcd/base/6_1_merged.gds results/nangate45/gcd/base/6_final.gds
Log                       Elapsed seconds
1_1_yosys                          2
3_3_place_gp                       1
4_1_cts                            8
5_2_route                   10
6_1_merge                         10
6_report                           3
```

### Environment

```shell
The script doesn't seem to work?

I have no name!@6c6a562a4655:/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh 
tee: build/openroad-env.log: Permission denied                                                                                                              
Unknown git commit, this is not a git repository.                                                                                                           
                                                                                                                                                            
Please make sure that you have the latest code changes and add the commit                                                                                   
hash in the description.                                                      
                                                                              
kernel: Linux 6.8.0-55-generic                                                
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)                      
cmake version 3.24.2                                                          
CMake Warning at CMakeLists.txt:98 (message):                                                                                                               
  OpenROAD git describe failed, using sha1 instead                     
                                                                                                                                                            
                                                                              
CMake Error: Could not open file for write in copy operation /OpenROAD-flow-scripts/tools/OpenROAD/include/ord/Version.hh.tmp                               
CMake Error: : System Error: Inappropriate ioctl for device         
CMake Error at CMakeLists.txt:135 (configure_file):   
  configure_file Problem configuring file                                                                                                                   
                                                                              
                                                                                                                                                            
-- The CXX compiler identification is GNU 11.4.0                                                                                                            
-- Detecting CXX compiler ABI info                                                                                                                          
-- Detecting CXX compiler ABI info - done                       

The `./build_openroad.sh` script produces the following output;

#21 [stage-2  7/11] COPY --link flow/scripts flow/scripts
#21 CACHED

#22 [stage-2  3/11] COPY --link docs docs
#22 CACHED

#23 [stage-2 11/11] COPY --link      --exclude=.git* --exclude=tools/ --exclude=docs/ --exclude=docker/      --exclude=flow/designs --exclude=flow/platforms --exclude=flow/scripts      --exclude=flow/test --exclude=flow/tutorials --exclude=flow/util   . ./
#23 CACHED

#24 exporting to image
#24 exporting layers done
#24 writing image sha256:35fabdf7989da7654b11f10e56ec0cc5fa69843b7afaaacea2e00b7f577211ab done
#24 naming to docker.io/openroad/flow-ubuntu22.04-builder:bc62c0 done
#24 DONE 0.0s
```

### To Reproduce

Follow the instructions @ https://openroad-flow-scripts.readthedocs.io/en/latest/tutorials/FlowTutorial.html#verifying-the-docker-based-installation

Got to the step;
```
docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22.04-builder
cd flow
make
```

Wasn't sure what issue command to use, so adding the output of `make synth_issue`

[synth_gcd_nangate45_base_2025-04-03_23-30.tar.gz](https://github.com/user-attachments/files/19596029/synth_gcd_nangate45_base_2025-04-03_23-30.tar.gz)

### Relevant log output

```shell

```

oharboe, 2025-04-06:
@mithro Just FYI. This is a slight non-sequitor, but if you have Bazelisk installed and you want to bring up the GUI without having to install anything more, run:

    cd flow
    bazelisk run designs/asap7/gcd:gcd_floorplan /tmp/floorplan gui_floorplan

maliberty, 2025-04-06:
@oharboe how do you get the GUI working when the bazel build of OR doesn't include qt yet?

oharboe, 2025-04-06:
> @oharboe how do you get the GUI working when the bazel build of OR doesn't include qt yet?

It automatically gets the docker image version for now. Later we will have qt in bazel 🤞

povik, 2025-04-07:
I can reproduce by:

    docker run --rm -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow:/OpenROAD-flow-scripts/flow openroad/flow-ubuntu22.04-builder

and then, inside the container:

    source ./env.sh
    make -C flow

The cause is that the Yosys version is old:

    $ yosys -V
    Yosys 0.40 (git sha1 UNKNOWN, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

@maliberty @vvbandeira Are we forgetting to update the Yosys version in the docker images? Is `openroad/flow-ubuntu22.04-builder` supposed to be current?

oharboe, 2025-04-07:
How about updating the tutorial to use latest Docker image? Building OpenROAD is a painful task that is a separate concern to learning the ORFS flow.

https://hub.docker.com/r/openroad/orfs/tags

mguthaus, 2025-04-10:
I think it's related to this bug I filed:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2760

maliberty, 2025-06-04:
@vvbandeira should we swap openroad/flow-ubuntu22.04-builder for openroad/flow-ubuntu22.04-dev in the tutorial?