The-OpenROAD-Project/OpenROAD-flow-scripts #3046: Non determinism in gf180/aes when runing STA in paralell and with 1 thread.

arthurjolo, 2025-04-04:
### Subject

[Documentation] for documentation errors.

### Describe the bug

The design gf180/aes shows different results when I add the command `sta::set_thread_count 1` to the script "flow/scripts/load.tcl" compared to running without the command.

Results for final tns without `sta::set_thread_count 1`:
```
==========================================================================
finish report_tns
--------------------------------------------------------------------------
tns -122.07

==========================================================================
finish report_wns
--------------------------------------------------------------------------
wns -0.95
```

Results for final tns with `sta::set_thread_count 1`:
```
==========================================================================
finish report_tns
--------------------------------------------------------------------------
tns -123.49

==========================================================================
finish report_wns
--------------------------------------------------------------------------
wns -0.94
```

### Expected Behavior

Have the same result running with `sta::set_thread_count 1`  and without

### Environment

```shell
Git commit: 57ca9a2d109fda9c0b2fd329c846bbc34931a36a
kernel: Linux 6.8.0-1021-gcp
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-20249-g57ca9a2d10
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
[... 356 lines cut ...]
                                  4.44   data arrival time

                          3.00    3.00   clock clk (rise edge)
                          0.00    3.00   clock source latency
     1    0.16    0.00    0.00    3.00 ^ clk (in)
                                         clk (net)
                  0.04    0.01    3.01 ^ clkbuf_0_clk/I (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
     4    0.40    0.16    0.17    3.18 ^ clkbuf_0_clk/Z (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
                                         clknet_0_clk (net)
                  0.16    0.01    3.19 ^ clkbuf_2_2_0_clk/I (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
     7    0.59    0.23    0.23    3.42 ^ clkbuf_2_2_0_clk/Z (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
                                         clknet_2_2_0_clk (net)
                  0.23    0.01    3.43 ^ clkbuf_leaf_23_clk/I (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
    22    0.19    0.09    0.17    3.60 ^ clkbuf_leaf_23_clk/Z (gf180mcu_fd_sc_mcu9t5v0__clkbuf_20)
                                         clknet_leaf_23_clk (net)
                  0.09    0.00    3.60 ^ sa13_sr[0]$_DFF_P_/CLK (gf180mcu_fd_sc_mcu9t5v0__dffq_4)
                          0.02    3.62   clock reconvergence pessimism
                         -0.11    3.51   library setup time
                                  3.51   data required time
-----------------------------------------------------------------------------
                                  3.51   data required time
                                 -4.44   data arrival time
-----------------------------------------------------------------------------
                                 -0.94   slack (VIOLATED)
```

maliberty, 2025-04-04:
@tspyrou please ask Cherry to take a look.

parallaxsw, 2025-04-24:
Error: cts_with_no_sta_threads.tcl, 4 incompatible database schema revision 0.107 > 0.106

maliberty, 2025-04-24:
You are probably using an old build of OR if you get that error.

tspyrou, 2025-04-25:
@arthurjolo Can you update the testcase to work with the odb at the head of openroad as of today. This will help @jjcherry56 efficiently get to this.

tspyrou, 2025-04-25:
@arthurjolo assign back to me when done.

tspyrou, 2025-04-25:
@arthurjolo when you rerun can you record the git hash for the commit you are on?

maliberty, 2025-04-25:
I tested this just a few days ago.  That error suggests the build Cherry is using is older and can't read a current db not the other way around.

parallaxsw, 2025-04-25:
Ya, I couldn't tell which was too old from the error message

tspyrou, 2025-04-28:
Cherry reports that TSAN is clean and that the diff is reliably reproduced at resizer iteration 112. He thinks it may be a floating point rounding difference.

tspyrou, 2025-04-29:
Latest update is that this is reproducible without any netlist editing.

arthurjolo, 2025-04-29:
Hi tom, it was confirmed that the issue is not in sta? Should someone in RSZ take a look at this?

maliberty, 2025-04-29:
@arthurjolo if it can be reproduced without any netlist editing then it is an sta issue.

jjcherry56, 2025-05-02:
7e5cf132 non-determinism in required times resolves orfs #3046

maliberty, 2025-05-02:
Thanks started a secure-CI on an sta update with this fix.

closed 2025-06-16 by arthurjolo