The-OpenROAD-Project/OpenROAD-flow-scripts #3086: LEF definition spacing between vvias on the same net

maliberty, 2025-04-16:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/3085

<div type='discussions-op-text'>

<sup>Originally posted by **pars97** April 16, 2025</sup>
I am using my own custom technology files that I created. 
In this kit, I must have a spacing of 2 micon (yes it's huge) between Via1 and Via2 layers.

(I'll add a .zip with the complete tech that you can run if you put it in flow/platform)

In my LEF describing the cells, I am pushing the information on where are the Via1 by using the OBS definition. Here is a short cut of a file : 

```
  OBS
      LAYER Via1 ;
        POLYGON 6.5 0.6 6.5 2.1 8.0 2.1 8.0 0.6 ;
        POLYGON 19.6 0.6 19.6 2.1 21.1 2.1 21.1 0.6 ;.
[...]
END
```
In my tech LEF definition, I have the following statements for Via1 and Via2 statements : 
```
LAYER Via1
  TYPE CUT ;
  SPACING 2 ;
  WIDTH   1.5 ;
  SPACING 2 SAMENET LAYER Via2 ;

  ENCLOSURE BELOW 0.00 0.06 ;
  ENCLOSURE ABOVE 0.01 0.06 ;
  PROPERTY LEF58_EOLENCLOSURE "
  	EOLENCLOSURE 0.34 0.06 ;" ; 

  ARRAYSPACING CUTSPACING 0.36 ARRAYCUTS 4 SPACING 0.36 ; # Vn.2b

  ACCURRENTDENSITY AVERAGE 0.28 ;
  DCCURRENTDENSITY AVERAGE 0.18 ;
  ANTENNAMODEL OXIDE1 ;
  ANTENNAAREARATIO 20.0 ;
END Via1
```

and 
```
LAYER Via2
  TYPE CUT ;
  PITCH 2 ;
  WIDTH   2 ;
  SPACING 2 ;
  SPACING 2 SAMENET LAYER Via1 ;

  ENCLOSURE BELOW 1.5 1.5 ;
  ENCLOSURE ABOVE 1.5 1.5 ;
  PROPERTY LEF58_EOLENCLOSURE "
  	EOLENCLOSURE 1.5 1.5 ;" ; 

  ACCURRENTDENSITY AVERAGE 0.28 ;
  DCCURRENTDENSITY AVERAGE 0.18 ;
  ANTENNAMODEL OXIDE1 ;
  ANTENNAAREARATIO 20.0 ;
END Via2
```

The ```SPACING 2 SAMENET LAYER Via1 ;``` is the statement describing the distance between the two layers. I repeteaded this statement for the Via1 description in the hopes it fizes my problem but it didnt. 

Most of the time, the router does it's job to avoid putting Via2 close to Via1, but in some instance, it fails. 
For example 
This is unnacceptable (X:126.8 Y:299) : 
[image]
And this is ok : 
[image]

Overall, I have around 30 mistakes over my circuit, which is not much but still a bit bizzare. 

Do you have any insight on how the .LEF are treated and what should I change to assure that they are spaced of 2um at all times?

In this .zip, I've replicated the OpenRoad folder tree and added my design plus my tech files. 
[flow.zip](https://github.com/user-attachments/files/19780249/flow.zip)

Let me know if anything else is required.
Many thanks.

</div>

maliberty, 2025-04-28:
@osamahammad21 suggestions?

pars97, 2025-05-02:
After trying multiple ways, I found that my tought process regarding the LEF definition was wrong.

The OBS LEF definition is there to mention to the router that there is an obstacle preventing the layer mentionned to be put it. Not that the obstacle is on said layer.

For instance,  
```
OBS
      LAYER Via1 ;
        POLYGON 6.5 0.6 6.5 2.1 8.0 2.1 8.0 0.6 ;
        POLYGON 19.6 0.6 19.6 2.1 21.1 2.1 21.1 0.6 ;.
[...]
END
``` 

means that the router cannot place the VIA1 layer on the polygons mentionned below. 

I changed all my LEF definitions macros for all my cell to correctly represent the issue at hand which was that VIA2 was where it was not supposed to be. Hence my LEF files now read:

```
OBS
      LAYER Via2 ;
        POLYGON 6.5 0.6 6.5 2.1 8.0 2.1 8.0 0.6 ;
        POLYGON 19.6 0.6 19.6 2.1 21.1 2.1 21.1 0.6 ;.
[...]
END
```

Effectively blocking the VIA2 over the VIA1 space as the polygon are the VIA1 polygons.

Doing so I have no more errors in VIA2-VIA1 distance and placement but a new issue has arisen, i have some MT2-MT2 spacing issue that were not there before.

[image]

The distance should be 2um and above. I've added the MACRO_PLACE_HALO ?= 2 2 command but I still have some issue with that. 

For now, this one can be considered closed as the original post was successfully completed although I would be gratefull for your input on the Metal-Metal distance issue.

maliberty, 2025-05-02:
I can't tell if these shapes are part of the macro or general wiring.  We should take account of spacing between the via and wire.

closed 2025-05-02 by maliberty

pars97, 2025-05-02:
The square purple bownish is defined by VIA_RULE GENERATE statement and the purple only is probably related to routing. But it seems the routing doesnt recognize correctly in this instance that the VIA does have an extra METAL2 enclosure that it must route around. 

Another instance of this is like this  :

[image]

I try to see what causes this issue because it's a very rare occurence (the only two occurences in my circuit at the moment).

pars97, 2025-05-16:
To finally close this. I think the problem was not caused by Openroad at all. There were some spacing mistakes in the cells themselves. I have resolved the cell issues and my designs are now error free.