The-OpenROAD-Project/OpenROAD-flow-scripts #3117: installation fails during build_openroad.sh

Cronus-38, 2025-04-29:
### Subject

[Build] for compilation-related issues.

### Describe the bug

The installation of the ORFS fails during execution of the build_openroad.sh script:

```
...
Setting up qt5-image-formats-plugins:amd64 (5.15.3-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
Errors were encountered while processing:
 ca-certificates
 liblwp-protocol-https-perl
 libwww-perl
 devscripts
E: Sub-process /usr/bin/dpkg returned an error code (1)
Error: building at STEP "RUN ./DependencyInstaller.sh -base $options $constantBuildDir     && ./DependencyInstaller.sh -common $options $constantBuildDir     && rm -rf /tmp/installer /tmp/* /var/tmp/* /var/lib/apt/lists/*": while running runtime: exit status 100
```

### Expected Behavior

Should install without any problems.

### Environment

```shell
That file doesn't exist since it didn't make it that far into the installation process.  But here's this, FWIW:

linux> lsb_release -a
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: RedHatEnterprise
Description:    Red Hat Enterprise Linux release 8.10 (Ootpa)
Release:        8.10
Codename:       Ootpa
```

### To Reproduce

linux> git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
> cd OpenROAD-flow-scripts
> ./build_openroad.sh

### Relevant log output

```shell

```

Cronus-38, 2025-05-08:
Hello, just checking in to see if you have any additional info on what may have changed.  Nothing has changed on our end, as far as I know.

maliberty, 2025-05-08:
We don't have official rhel8 support yet.  I don't think we have changed anything material but we also don't test on that platform.

Cronus-38, 2025-05-14:
I collaborated with a colleague to identify and resolve the issue we were facing. 

We traced the problem to the _installORDependencies function, which executes the OpenROAD tool's DependencyInstaller.sh. (Note that the missing dependency is not related to the DependencyInstaller.sh file associated with the OpenROAD-flow-scripts, rather the problem lies in the DependencyInstaller.sh for the OpenROAD tool itself).  During execution of build_openroad.sh we encountered the missing dependency described in this issue.

We resolved the issue by adding libssl-dev to the list of dependencies in the _installUbuntuPackages() function within OpenROAD's DependencyInstaller.sh. As a result, we are now successfully running the Docker image on RHEL8.

In our internal discussions, we considered the possibility of adding this dependency to the ORFS DependencyInstaller.sh. However, we tested and confirmed that modifying OpenROAD's DependencyInstaller.sh solves the problem.  As for whether or not this is the optimal fix, we are open to feedback.  We'd be happy to discuss it further if the above explanation isn't clear.

maliberty, 2025-05-14:
We plan to add rhel8 to our supported platforms shortly so this is helpful. I can't test such a pr but would be ok to take it on your experience.

closed 2025-08-28 by vvbandeira