The-OpenROAD-Project/OpenROAD-flow-scripts #3129: Broken KLayout URL fails Docker build

erendn, 2025-05-04:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Similar to issue https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1099, the Docker build fails because the current KLayout version was removed from the website.

I tried bumping to the latest version in DependencyInstaller.sh but it still doesn't compile, saying some dependencies are missing. Can someone take a look?

### Expected Behavior

`./build_openroad.sh` should build OpenROAD and all its dependencies.

### Environment

```shell
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.4.0-186-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.16.3
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-21009-ga8ef34ce2
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds                                                                                                                                 -- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull                                                                                                                                      -- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized                                                                                                                            -- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow                                                                                                                       [44/4616]-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format - Success
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR
  GTEST_MAIN_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindGTest.cmake:263 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/CMakeLists.txt:142 (find_package)

-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17 - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17 - Success
-- Performing Test C_COMPILER_SUPPORTS__-fno-exceptions
-- Performing Test C_COMPILER_SUPPORTS__-fno-exceptions - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-fno-exceptions
-- Performing Test CXX_COMPILER_SUPPORTS__-fno-exceptions - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.1", minimum required is "4.0")
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")
-- boost: 1.71.0
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Clone the repo and run `build_openroad.sh`.

### Relevant log output

```shell

```

young03600, 2025-05-06:
Same error when building the ORFS using docker image
Any update?

maliberty, 2025-05-06:
Just merged https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/3131 - please try again with the head.

young03600, 2025-05-06:
Thanks for the update 
But I got the error

```
#21 240.1 mkdir -p /OpenROAD-flow-scripts/tools/install/yosys/bin
#21 240.1 cp yosys-abc yosys yosys-config yosys-filterlib yosys-smtbmc yosys-witness /OpenROAD-flow-scripts/tools/install/yosys/bin
#21 240.3 strip -S /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
#21 240.4 strip /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
#21 240.5 strip /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
#21 240.5 mkdir -p /OpenROAD-flow-scripts/tools/install/yosys/share/yosys
#21 240.5 cp -r share/. /OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
#21 240.5 make: Leaving directory '/OpenROAD-flow-scripts/tools/yosys'
#21 240.5 [INFO FLW-0030] Compiling yosys-slang.
#21 240.5 make: Entering directory '/OpenROAD-flow-scripts/tools/yosys-slang'
#21 240.5 cmake -S . -B build -DYOSYS_CONFIG=/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-config -DCMAKE_BUILD_TYPE=Release
#21 240.5 -- The CXX compiler identification is GNU 11.4.0
#21 240.5 -- Detecting CXX compiler ABI info
#21 240.6 -- Detecting CXX compiler ABI info - done
#21 240.6 -- Check for working CXX compiler: /usr/bin/c++ - skipped
#21 240.6 -- Detecting CXX compile features
#21 240.6 -- Detecting CXX compile features - done
#21 240.6 -- slang version: 8.0.0+0
#21 240.6 -- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
#21 240.6 -- Build STATIC library as: svlang
#21 240.7 -- {fmt} version: 11.1.5
#21 240.7 -- Build type: Release
#21 240.7 -- Using remote fmt library
#21 240.7 -- Using vendored boost_unordered header
#21 240.7 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
#21 240.7 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
#21 240.7 -- Found Threads: TRUE  
#21 240.7 -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
#21 240.8 -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
#21 240.8 -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
#21 240.8 -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
#21 240.8 -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
#21 240.9 -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
#21 240.9 -- Using yosys: /OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-config
#21 240.9 -- yosys-config --datdir: /OpenROAD-flow-scripts/tools/install/yosys/share/yosys
#21 240.9 -- yosys-config --cxxflags (filtered): -D__TIME__=0;-D__DATE__=0;-D__TIMESTAMP__=0;-I/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/include;-D_YOSYS_;-I/OpenROAD-flow-scripts/tools/install/yosys/include;-DYOSYS_VER="0.51";-DYOSYS_MAJOR=0;-DYOSYS_MINOR=51;-DYOSYS_COMMIT=0.51;-DYOSYS_ENABLE_READLINE;-DYOSYS_ENABLE_PLUGINS;-DYOSYS_ENABLE_GLOB;-DYOSYS_ENABLE_ZLIB;-I/usr/include/tcl8.6;-DYOSYS_ENABLE_TCL;-DYOSYS_ENABLE_ABC;-DYOSYS_ENABLE_COVER
#21 240.9 fatal: not a git repository: /OpenROAD-flow-scripts/tools/yosys-slang/../../.git/modules/tools/yosys-slang
#21 240.9 CMake Error at cmake/GitRevision.cmake:4 (execute_process):
#21 240.9   execute_process failed command indexes:
#21 240.9 
#21 240.9     1: "Child return code: 128"
#21 240.9 
#21 240.9 Call Stack (most recent call first):
#21 240.9   src/CMakeLists.txt:4 (git_rev_parse)
#21 240.9 
#21 240.9 
#21 240.9 -- Configuring incomplete, errors occurred!
#21 240.9 See also "/OpenROAD-flow-scripts/tools/yosys-slang/build/CMakeFiles/CMakeOutput.log".
#21 240.9 make: *** [Makefile:18: configure] Error 1
#21 240.9 make: Leaving directory '/OpenROAD-flow-scripts/tools/yosys-slang'
#21 ERROR: process "/bin/sh -c echo \"\" > tools/yosys/abc/.gitcommit &&   env CFLAGS=\"-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined\"   CXXFLAGS=\"-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined\"   ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2
------
 > [orfs-builder-base 2/2] RUN echo "" > tools/yosys/abc/.gitcommit &&   env CFLAGS="-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined"   CXXFLAGS="-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined"   ./build_openroad.sh --no_init --local --threads 16:
240.9     1: "Child return code: 128"
240.9 
240.9 Call Stack (most recent call first):
240.9   src/CMakeLists.txt:4 (git_rev_parse)
240.9 
240.9 
240.9 -- Configuring incomplete, errors occurred!
240.9 See also "/OpenROAD-flow-scripts/tools/yosys-slang/build/CMakeFiles/CMakeOutput.log".
240.9 make: *** [Makefile:18: configure] Error 1
240.9 make: Leaving directory '/OpenROAD-flow-scripts/tools/yosys-slang'
------
ERROR: failed to solve: process "/bin/sh -c echo \"\" > tools/yosys/abc/.gitcommit &&   env CFLAGS=\"-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined\"   CXXFLAGS=\"-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined\"   ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2
```

maliberty, 2025-05-06:
No relation to klayout so closing this issue.  My guess is you need a `git submodule update --init --recursive` as yosys-slang is a new submodule.

closed 2025-05-06 by maliberty