The-OpenROAD-Project/OpenROAD-flow-scripts #3150: ERROR: RuntimeError: Macro not found in LEF file: DFFHQNx1_ASAP7_75t_R (line=69, cell=, file=6_final.def) in Layout.read

oharboe, 2025-05-08:
### Subject

[Documentation] for documentation errors.

### Describe the bug

untar and run [klayout-crash.tar.gz](https://github.com/user-attachments/files/20101158/klayout-crash.tar.gz)

```
OpenROAD v2.0-21011-g525ebd5bab 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
source external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/setRC.tcl
Deleted 0 routing obstructions
[INFO RCX-0431] Defined process_corner X with ext_model_index 0
[INFO RCX-0029] Defined extraction corner X
[INFO RCX-0435] Reading extraction model file external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/rcx_patterns.rules ...
[INFO RCX-0436] RC segment generation lb_32x128 (max_merge_res 50.0) ...
[INFO RCX-0040] Final 124 rc segments
[INFO RCX-0439] Coupling Cap extraction lb_32x128 ...
[INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
[INFO RCX-0442] 72% of 196 wires extracted
[INFO RCX-0442] 100% of 196 wires extracted
[INFO RCX-0045] Extract 57 nets, 165 rsegs, 165 caps, 95 ccs
[INFO RCX-0443] 57 nets finished
IR drop analysis for power nets is skipped because PWR_NETS_VOLTAGES is undefined
IR drop analysis for ground nets is skipped because GND_NETS_VOLTAGES is undefined
Cell type report:                       Count       Area
  Tie cell                                  4       0.17
  Buffer                                    3       0.22
  Clock buffer                              3       0.22
  Timing Repair Buffer                      9       0.85
  Inverter                                  4       0.17
  Sequential cell                           4       1.17
  Multi-Input combinational cell            8       0.70
  Total                                    35       3.50
[WARNING GUI-0076] QSocketNotifier: Can only be used with threads started with QThread
[INFO] Reporting cells prior to loading DEF ...
	external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds
Warning: DEF UNITS does not match reader DBU (DEF UNITS is 1000 and corresponds to a DBU of 0.001, but reader DBU is set to 0.00025) (line=5, cell=, file=6_final.def)
[INFO] Reporting cells prior to loading DEF ...
ERROR: RuntimeError: Macro not found in LEF file: DFFHQNx1_ASAP7_75t_R (line=69, cell=, file=6_final.def) in Layout.read
  external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/util/def2stream.py:23

```

### Expected Behavior

No error

### Environment

```shell
ORFS v3.0-3061-g9929f4e9
```

### To Reproduce

See above

### Relevant log output

```shell

```

jeffng-or, 2025-05-08:
Ran into some weird bazel-related issues with the testcase when running on a new machine with klayout 0.28.17:

- test case was looking for LEF file in /home//oyvind/.cache/bazel/_bazel_oyvind/6e078b44989a9ce1500a63165a20fd2a/external/_main~orfs_repositories~docker_orfs/OpenROAD-flow-scripts/flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
- asap7 platform gds file wasn't bundled in with the external directory

In any case, I was able to work around these and verify that klayout 0.28.17 completes successfully.

Will send up PR to go from 0.30.1 to 0.28.17 (note that both need libgit2-1.1, so the libgit2-dev dependency will need to remain in the DependencyInstaller)

jeffng-or, 2025-05-13:
> thanks!

@oharboe, did the change to KLayout 0.28.17 resolve the issue?

closed 2025-05-13 by jeffng-or