The-OpenROAD-Project/OpenROAD-flow-scripts #3177: need PRE_STEP_TCL variables to be implemented

Cronus-38, 2025-05-21:
### Description

Currently there are a number of POST_STEP_TCL variables available in ORFS, including POST_FLOORPLAN_TCL and POST_CTS_TCL.  These variables enable the user to source custom scripts at various points in the flow.  It would be very helpful to have similar "PRE_" variables available throughout the flow.

### Suggested Solution

As an example, currently in order to redefine a procedure that is called during execution of the cts.tcl script I need to either hack the cts.tcl script itself or hack the load.tcl script (which contains the pre-defined procedure I'm trying to redefine).  It would be a much more elegant solution to have a PRE_CTS_TCL variable which points to a script that gets sourced _after_ the load.tcl file is sourced at the beginning of the cts.tcl script.  This would avoid the need to hack anything in the default ORFS installation.  Like this:

if { [env_var_exists_and_non_empty PRE_CTS_TCL] } {
  source $::env(PRE_CTS_TCL)
}

It would be nice, too, if every step in the flow supported a POST_STEP_TCL variable (only a few do currently).

maliberty, 2025-05-22:
These hooks have been added as the need arose.  Feel free to make a PR with a more systematic treatment.

Dhirenderchoudhary, 2026-03-03:
@maliberty Can i do that ?

I have been looking for issues that are open from long time and not resolved