The-OpenROAD-Project/OpenROAD-flow-scripts #3180: Threading related crash in STA/CTS

jeffng-or, 2025-05-22:
### Subject

[Stage]: CTS.

### Describe the bug

Thread related crash in STA while in CTS step.

crash is not seen if sta::set_thread_count 1 is set.

I have a make issue tar file that reproduces the problem, along with TSAN output. I've shared the data via Google Drive with the appropriate developers. Please contact me if you think you need access. The TSAN output is in a file called output.log.

### Expected Behavior

No crashes

### Environment

```shell
Ran issue tarfile with TSAN-enabled build of ORFS commit tag cc477d8a3d57a4872fb0cf32fa6ded41a79bbb1d
```

### To Reproduce

Unpack issue tar file found
cd into directory
Run ./run-me-cusack-asap7-base.sh

### Relevant log output

```shell
/usr/include/c++/11/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = sta::TableBlock<sta::Edge>*; _Alloc = std::allocator<sta::TableBlock<sta::Edge>*>; std::vector<_Tp, _Alloc>::const_reference = sta::TableBlock<sta::Edge>* const&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x0000621E1FF16452 in openroad
 1# 0x000079AA5E042520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x0000621E1FF0E978 in openroad
 6# 0x0000621E201BE653 in openroad
 7# sta::Path::vertex(sta::StaState const*) const in openroad
 8# sta::CheckCrpr::findCrpr(sta::Path const*, sta::Path const*, bool, float&, sta::Pin*&) in openroad
 9# sta::CheckCrpr::checkCrpr(sta::Path const*, sta::Path const*) in openroad
10# sta::Latches::latchBorrowInfo(sta::Path const*, sta::Path const*, sta::Path const*, float const&, bool, float&, float&, float&, float&, float&, float&, float&, bool&) const in openroad
11# sta::Latches::latchRequired(sta::Path const*, sta::Path const*, sta::Path const*, sta::MultiCyclePath const*, sta::PathDelay const*, float, float const&, float&, float&, float&, float&) const in openroad
12# sta::Latches::latchRequired(sta::Path const*, sta::Path const*, sta::Path const*, sta::PathAnalysisPt const*, float&, float&, float&, float&) const in openroad
13# sta::Latches::latchOutArrival(sta::Path const*, sta::TimingArc const*, sta::Edge const*, sta::PathAnalysisPt const*, sta::Tag*&, float&, float&) in openroad
14# sta::PathVisitor::visitFromPath(sta::Pin const*, sta::Vertex*, sta::RiseFall const*, sta::Path*, sta::Edge*, sta::TimingArc*, sta::Pin const*, sta::Vertex*, sta::RiseFall const*, sta::MinMax const*, sta::PathAnalysisPt const*) in openroad
15# sta::PathVisitor::visitEdge(sta::Pin const*, sta::Vertex*, sta::Edge*, sta::Pin const*, sta::Vertex*) in openroad
16# sta::PathVisitor::visitFaninPaths(sta::Vertex*) in openroad
17# sta::ArrivalVisitor::visit(sta::Vertex*) in openroad
18# 0x0000621E2027BBF6 in openroad
19# sta::DispatchQueue::dispatch_thread_handler(unsigned long) in openroad
20# 0x000079AA5E4DC253 in /lib/x86_64-linux-gnu/libstdc++.so.6
21# 0x000079AA5E094AC3 in /lib/x86_64-linux-gnu/libc.so.6
22# 0x000079AA5E126850 in /lib/x86_64-linux-gnu/libc.so.6
```

maliberty, 2025-06-06:
I'll start an sta update CI

maliberty, 2025-06-06:
Fixed https://github.com/The-OpenROAD-Project/OpenROAD/pull/7529

closed 2025-06-06 by maliberty