The-OpenROAD-Project/OpenROAD-flow-scripts #3264: Fill for non-metal/non-routing layers

stockstephanie, 2025-06-25:
### Description

Some PDKs have DRC requirements for density on layers besides routing metal. One I am using currently has pretty tight range requirements for a poly layer (29-30%) and I really liked the simplicity of fill through OpenROAD. Since this poly layer is not declared as a routing layer in the LEF, it cannot be assigned in the set_rc file, and then fill run on that layer.

### Suggested Solution

Not sure how difficult it would be to implement this or the nitty gritty behind it, but would it be possible to ignore fill layers not assigned in the set_rc file? Then a warning could be issued from these fill layers not being included in RC extraction if that is the reason behind fill failing. Thanks!

maliberty, 2025-06-26:
There is no restriction in the code that I see related to routing layers.  If you provide a configuration for the poly layer it should fill it.

stockstephanie, 2025-06-26:
Well, now that I've tried again I'm not getting the setRC error anymore that led me down my original train of thought. No idea how I managed that earlier. You're totally right.  I must have had something messed up originally because I had deleted the fill configuration for that layer when I was checking the metal fill rules, but now that it is back in it actually adds it. I apologize for that oversight.