The-OpenROAD-Project/OpenROAD-flow-scripts #3357: CTS / resizer hits max buffer count limit in hold violations with nangate45 bp_quad

jeffng-or, 2025-07-24:
### Subject

[Stage]: CTS.

### Describe the bug

Trying to align nangate45 bp_quad with the original source repo, but running into issue with CTS / Resizer. The resizer call post CTS is trying to deal with 226k hold violations which leads to hitting the max buffer count limit.

There are multiple clocks and I dumped the ODB before the resizer call. I didn't see anything out of the ordinary.

### Expected Behavior

passing run

### Environment

```shell
0ad7848b81b48934830b4d4da7073882f40885e0 - git commit tag

Update flow/designs/nangate45/bp_quad/bsg_chip.sdc
set clk_period 1.3
set clk_uncertainty 0.150
```

### To Reproduce

cts make issue tarball: https://drive.google.com/file/d/1SRk8mlSNirREZD5iN9AIuZ52YkfBKv7L/view?usp=drive_link

### Relevant log output

```shell
[INFO RSZ-0046] Found 226672 endpoints with hold violations.
Iteration | Resized | Buffers | Cloned Gates |   Area   |   WNS   |   TNS   | Endpoint
--------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |    +0.0% |  -1.221 | -7544.551 | next.downlink.ch_0__downstream.baf.bapg_wr.ptr_sync.sync_p_z_blss.bsg_SYNC_1_r_3_sv2v_reg$_DFF_P_/D
       10 |       0 |      10 |            0 |    +0.0% |  -1.194 | -7544.021 | next.downlink.ch_0__downstream.baf.bapg_wr.ptr_sync.sync_p_z_blss.bsg_SYNC_1_r_4_sv2v_reg$_DFF_P_/D
       20 |       0 |      22 |            0 |    +0.0% |  -1.194 | -7543.487 | next.downlink.ch_0__downstream.baf.bapg_wr.ptr_sync.sync_p_z_blss.bsg_SYNC_1_r_4_sv2v_reg$_DFF_P_/D
...
   159640 |       0 |  280825 |            0 |    +4.2% |  -1.190 | -2288.928 | next.downlink.ch_0__downstream.baf.bapg_wr.ptr_sync.sync_p_z_blss.bsg_SYNC_1_r_6_sv2v_reg$_DFF_P_/D
    final |       0 |  280831 |            0 |    +4.2% |  -1.190 | -2288.836 | next.downlink.ch_0__downstream.baf.bapg_wr.ptr_sync.sync_p_z_blss.bsg_SYNC_1_r_6_sv2v_reg$_DFF_P_/D
--------------------------------------------------------------------------------------
[WARNING RSZ-0066] Unable to repair all hold violations.
[INFO RSZ-0032] Inserted 280831 hold buffers.
[ERROR RSZ-0060] Max buffer count reached.
```

maliberty, 2025-07-24:
@jeffng-or does this seem to be a cts issue or a rsz one based on your investigation?

jeffng-or, 2025-07-24:
> [@jeffng-or](https://github.com/jeffng-or) does this seem to be a cts issue or a rsz one based on your investigation?

The clock trees looked fairly balanced with 0.146ns as the skew for the worst case path, so maybe rsz? The slack for that path is -1.83ns on a 1.3ns clock.

The ODB that was saved right after CTS is called 4_1_cts-post.odb.

maliberty, 2025-09-04:
You can have paths cross between clocks though often there is some sort of synchronization logic involved to prevent timing issues.   I don't know much about this chip's design to be sure how it was handled.  Do you see anything interesting along the path?

arthurjolo, 2025-09-04:
I see no synchronization, what I mentioned that I believed it was another register on the path was just a clk_buff that I got confused. 

The worse paths are just one register from one clock connected to a register from the other clock.

[image]

Others have some logic with ANDs and AOIs

[image]

rovinski, 2025-09-06:
Is `To` connected directly to another register?

If so, then that is indeed a 2-FF synchronizer which is not properly constrained in SDC. Clock domain crossings are always* counted as hold violations, and there is nothing CTS or RSZ can do to fix that. There needs to be a `set_false_path` from `From` to `To` to remove the false hold violation. Additionally, there needs to be a constraint on the FF that `To` connects to. There are several ways to do this in SDC, unfortunately none of them are great. The [best approach I've seen](https://gist.github.com/brabect1/7695ead3d79be47576890bbcd61fe426) uses `set_max_delay` from the first synchronizer stage (`To`) to the second synchronizer stage (FF after `To`) and sets it to a "very small value" which is unfortunately dependent on the process.

In an ideal world, the PDK has a synchronizer cell which is just 2 FFs chained together and avoids needing to set timing constraints between them. But this also isn't portable RTL.

If there is logic after `To` (not just buffers/inverters), then the RTL is kind of broken. But from my familiarity with the BP source RTL, I think all of the CDCs are synchronized.

*unless one of them is generated from the other, which is not the case here.

arthurjolo, 2025-09-22:
@rovinski `To` is connected directly to another register. So This seems to a be a synchronizers, they even have "sync" on their name 😄 

However on the other cases where such as the one on the second image, where there is logic between the register form different clocks, the output of `To` is connected to other logic gates. Dan Petrisko commented that there can by asynchronous CDC FIFOs, I am checking to see if that is the case.