The-OpenROAD-Project/OpenROAD-flow-scripts #3420: AutoTuner - sweep mode assumes that the start directory is writable

jeffng-or, 2025-08-21:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

When using sweep mode, the AutoTuner writes the SDC file to the current directory, which might not be writable. If running within Docker with a starting directory of /OpenROAD-flow-scripts/tools/AutoTuner/src that is part of the image, the directory is not writable. Might want to be writing the file to somewhere under ORFS/flow instead.

### Expected Behavior

Write the file to a writable location

### Environment

```shell
aad91813f6449c7015dc05fa71156bb3da2ed9d2 - git commit tag
```

### To Reproduce

Build docker image with:
```
# syntax=docker/dockerfile:1
#
# Installs ORFS from docker image 
#

FROM openroad/orfs:v3.0-3615-ge01db632

# install AT required packages
RUN pip3 install -U -r /OpenROAD-flow-scripts/tools/AutoTuner/requirements.txt

# ORFS installation dir
WORKDIR /OpenROAD-flow-scripts/tools/AutoTuner/src
```
In docker:
```
python3 -m autotuner.distributed --design gcd --platform asap7 --config /OpenROAD-flow-scripts/flow/designs/asap7/gcd/autotuner.json sweep
```

### Relevant log output

```shell
I have no name!@a79f50fb145a:/OpenROAD-flow-scripts/tools/AutoTuner/src$ python3 -m autotuner.distributed --design gcd --platform asap7 --config /OpenROAD-flow-scripts/flow/designs/asap7/gcd/autotuner_freq.json sweep
/OpenROAD-flow-scripts/flow/designs/asap7/gcd/constraint.sdc
[INFO TUN-0012] Log folder /OpenROAD-flow-scripts/flow/logs/asap7/gcd.
2025-08-21 16:05:48,784	WARNING services.py:2072 -- WARNING: The object store is using /tmp instead of /dev/shm because /dev/shm has only 67108864 bytes available. This will harm performance! You may be able to free up space by deleting files in /dev/shm. If you are inside a Docker container, you can increase /dev/shm size by passing '--shm-size=10.24gb' to 'docker run' (or add it to the run_options list in a Ray cluster config). Make sure to set this to more than 30% of available RAM.
2025-08-21 16:05:48,845	INFO worker.py:1832 -- Started a local Ray instance. View the dashboard at 127.0.0.1:8265 
[INFO TUN-0009] Waiting for results.
(consumer pid=383) [INFO TUN-0007] Scheduling run for parameter /OpenROAD-flow-scripts.
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/distributed.py", line 649, in <module>
    main()
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/distributed.py", line 645, in main
    sweep()
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/distributed.py", line 580, in sweep
    ray.get(workers)
  File "/usr/local/lib/python3.10/dist-packages/ray/_private/auto_init_hook.py", line 21, in auto_init_wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/ray/_private/client_mode_hook.py", line 103, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/ray/_private/worker.py", line 2771, in get
    values, debugger_breakpoint = worker.get_objects(object_refs, timeout=timeout)
  File "/usr/local/lib/python3.10/dist-packages/ray/_private/worker.py", line 919, in get_objects
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(PermissionError): ray::consumer() (pid=383, ip=172.17.0.2)
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/utils.py", line 655, in consumer
    ray.get(openroad_distributed.remote(*next_item))
ray.exceptions.RayTaskError(PermissionError): ray::openroad_distributed() (pid=386, ip=172.17.0.2)
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/utils.py", line 625, in openroad_distributed
    config = parse_config(
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/utils.py", line 257, in parse_config
    write_sdc(sdc, path, sdc_original, constraints_sdc)
  File "/OpenROAD-flow-scripts/tools/AutoTuner/src/autotuner/utils.py", line 116, in write_sdc
    with open(file_name, "w") as file:
PermissionError: [Errno 13] Permission denied: '/OpenROAD-flow-scripts/tools/AutoTuner/src/constraint.sdc'
(consumer pid=397) [INFO TUN-0007] Scheduling run for parameter /OpenROAD-flow-scripts. [repeated 2x across cluster] (Ray deduplicates logs by default. Set RAY_DEDUP_LOGS=0 to disable log deduplication, or see https://docs.ray.io/en/master/ray-observability/user-guides/configure-logging.html#log-deduplication for more options.)
```

jeffng-or, 2025-09-19:
Requesting bumping up the priority.

Rapidus would like to start including sweeps in the demo. I'm working around this by running tune, but tune is also running the same config more than once (i.e. waste of a trial). In a recent run of 100 trials, almost 30% of the runs were duplicates.

Bonus points if sweep mode can write out data so that it can be viewable in Tensorboard like tune.

vvbandeira, 2025-11-26:
Please re-open here or upstream if this still needs attention.