The-OpenROAD-Project/OpenROAD-flow-scripts #3439: ihp-sg13g2: congestion violations around InOut I/O pads

Syndace, 2025-08-27:
### Subject

[Stage]: Global Router.

### Describe the bug

When using InOut I/O pads with the ihp-sg13g2 PDK, global routing fails with GRT-0116. Viewing the congestion report in the DRC viewer reveals that congestion is weirdly reported around the InOut cells, bondpads and even nearby fillers.

### Expected Behavior

.

### Environment

```shell
tools/OpenROAD doesn't exist in the Docker container and the script gives an error when ran on the host. I am running the newest master commit in the Docker container though, so the environment should be up-to-date and reproducible.
```

### To Reproduce

Here's the output of `make global_route_issue`. I have removed some of the .odb's for earlier steps from the reports directory, since the file size was too large to upload on GitHub. I can try to build a smaller example if the build times make debugging hard, just let me know :)

[global_route_cavaboris_ihp-sg13g2_base_2025-08-27_09-50.tar.gz](https://github.com/user-attachments/files/22004905/global_route_cavaboris_ihp-sg13g2_base_2025-08-27_09-50.tar.gz)

### Relevant log output

```shell
[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
Metal1               0             0            0.00%             0 /  0 /  0
Metal2          776715        150476           19.37%             1 /  0 / 87
Metal3          648301        165502           25.53%             0 /  1 / 168
Metal4          794939           919            0.12%             1 /  0 / 21
Metal5           47758          3044            6.37%             0 /  1 /  3
---------------------------------------------------------------------------------------
Total          2267713        319941           14.11%             2 /  2 / 279

[INFO GRT-0018] Total wirelength: 3389652 um
[INFO GRT-0014] Routed nets: 61132
[ERROR GRT-0116] Global routing finished with congestion. Check the congestion regions in the DRC Viewer.
Error: global_route.tcl, 113 GRT-0116
```

### Screenshots

[image]

[image]

### Additional Context

This seems very similar to https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2656, which has been fixed, except I get both vertical and horizontal violations. Unfortunately, I couldn't find any obvious difference between the `i2c-gpio-expander` and my design, so I don't know why the example does not have this issue.

Syndace, 2025-08-30:
Okay, I found the issue. I used `PRE_GLOBAL_ROUTE_TCL` with a tcl calling `write_verilog` to inspect the netlist just before global routing starts and found the following:

```Verilog
 sg13g2_dlygate4sd3_1 hold38456 (.A(p_inout_hyperbus_rwds),
    .X(net38456));

 ...

 sg13g2_IOPadInOut4mA u_pad_io_hyperbus_rwds (.c2p(net33620),
    .c2p_en(net35484),
    .p2c(core_io_hyperbus_rwds_read),
    .pad(net38456));
```

So what happens is, the flow inserts a hold buffer between the pad and the top-level, which leads to the weird routing from the bondpad back to the hold buffer in the core area. This is suspicious too since the direction of `p_inout_hyperbus_rwds` is `inout`, `sg13g2_IOPadInOut4mA.pad` is `INOUT` in the LEF, while `sg13g2_dlygate4sd3_1.A` is just `INPUT`. So this hold buffer not only causes routing problems, but also potentially prevents the output capability of the pad from working at all.

I've tried preventing buffers to be inserted into those nets by pointing `IO_CONSTRAINTS` to a tcl file containing `set_dont_touch [get_nets p_*]`, but `repair_timing` then errors out when trying to add the hold buffer:

```
[ERROR ODB-0373] Attempt to connect iterm to dont_touch net p_inout_hyperbus_dq[2]
```

I see two potential problems here:
- unidirectional hold buffers should not be inserted between `inout` connections
- `repair_timing` should not fail when trying to insert a buffer in a `dont_touch` net, but search for a different place to insert it instead

What do you think?

rovinski, 2025-08-31:
Are the bondpads abutting the corresponding pins on the I/O cells? If not, then the bondpad will not be connected to the pad cell and will require RDL routing, not regular routing. Normally the `PAD` tool marks these nets as `SPECIAL` so that they do not get touched by optimization or the detailed router, but if they aren't connected by abutment and there is no RDL routing, then it might be that `PAD` doesn't mark it as special.

If the bondpads _are_ connected by abutment (I'm just looking at the picture, I can't tell), it would be worth investigating why the net is being modified because I believe all tools are supposed to skip nets connected by abutment.

Syndace, 2025-08-31:
> Are the bondpads abutting the corresponding pins on the I/O cells?

How can I check that? Is there an indicator in the GUI?

In my footprint.tcl (attached to an earlier comment as well), I place the bondpads first and then run `connect_by_abutment` as the last command before removing I/O rows and finishing placement. The end of the script looks like this:

```tcl
...

# Place corners
place_corners sg13g2_Corner

# Place IO fill
set iofill {sg13g2_Filler10000
            sg13g2_Filler4000
            sg13g2_Filler2000
            sg13g2_Filler1000
            sg13g2_Filler400
            sg13g2_Filler200} ;
place_io_fill -row IO_NORTH {*}$iofill
place_io_fill -row IO_SOUTH {*}$iofill
place_io_fill -row IO_WEST {*}$iofill
place_io_fill -row IO_EAST {*}$iofill

# Place the bondpads
place_bondpad -bond bondpad_70x70 u_pad_* -offset "5.0 -$BONDPAD_SIZE.0"

# Connect ring signals
connect_by_abutment

remove_io_rows
```

Syndace, 2025-08-31:
Okay, from what I can tell, both should be given. In the GUI, I disabled almost everything but `Misc->Instances->Pins`:

[image]

This confirms there are shapes on the PIN layer, right?

The bounding boxes confirm the shapes are touching too:

```
Shape on padcell: (1292, 140), (1362, 143)
Shape on bondpad: (1292, 70), (1362, 140)
```

For reference, here is the [bondpad LEF](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/ihp-sg13g2/lef/bondpad_70x70.lef) and [the LEF containing the padcell `sg13g2_IOPadInOut4mA`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/ihp-sg13g2/lef/sg13g2_io.lef).

rovinski, 2025-08-31:
Yep, looks abutted to me.

> does it have to be _only_ `PIN`? Because there are shapes on `OBS` in addition to those on `PIN`.

No, `PIN` overrides `OBS`. The `OBS` is actually unnecessary in that LEF.

closed 2025-09-02 by maliberty