The-OpenROAD-Project/OpenROAD-flow-scripts #3446: @maliberty @vvbandeira @povik  This breaks bazel-orfs because OPENROAD_EXE is not a dependency for Yosys canonicalization, the ranslation of Verilog & SystemVerilog to .rtilil step:

oharboe, 2025-08-28:
@maliberty @vvbandeira @povik  This breaks bazel-orfs because OPENROAD_EXE is not a dependency for Yosys canonicalization, the ranslation of Verilog & SystemVerilog to .rtilil step:

```
$ bazelisk build BoomTile_grt
INFO: Invocation ID: aa6e75d9-c9ff-4c3d-83a7-6cdbf52f5d8c
INFO: Analyzed target //:BoomTile_grt (1 packages loaded, 5 targets configured).
ERROR: /home/oyvind/megaboom/BUILD.bazel:196:14: Action results/asap7/btb_128x56/base/1_1_yosys_canonicalize.rtlil failed: (Exit 2): bash failed: error executing Action command (from target //:btb_128x56_synth) /bin/bash -c ... (remaining 6 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/flow/scripts/variables.mk:102: *** OPENROAD_EXE is set to '/home/oyvind/.cache/bazel/_bazel_oyvind/7e6ad621f3f951c3ee6f5b179289b54e/sandbox/processwrapper-sandbox/29/execroot/_main/external/bazel-orfs++orfs_repositories+docker_orfs/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad', but it is either not found or not executable..  Stop.
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/3430#issuecomment-3232153075_

oharboe, 2025-08-28:
> What if we add three (or more) rules named, e.g., `check-openroad`, `check-yosys`, `check-klayout`, and put them as the first dependency for each flow stage/major rule in the Makefile? Would that work for Bazel?

I think this is asking for trouble, I think we should only execute this checking code if the user explicitly ask for it, not in the normal flow.

> I found the issue with the `versions.txt` file because I was looking for a fast way to make sure that when running AT, we had everything in place to complete the flow. It happened to me more than once while testing new features in AT that a run would start, go through the synthesis, and then fail because it did not have OR set up correctly. Then, after fixing the setup, I would have to restart the experiment from synthesis again. It would be nice to have an early check and error out early rather than later.

How about moving these checks into the versions.txt target?

Possibly add check-XXX  targets, but don't invoke them from the normal flow.

maliberty, 2025-08-28:
Nothing to do with AT per se - mostly first time ORFS users.  I'm not sure what a "non-make use-case" is.

oharboe, 2025-08-28:
Suggestion: we run whatever checks we think are useful when a user/script uses make, but that it isn't used required by bazel-orfs usecase with the "do-" targets, or when running "make print-FOO", etc.

closed 2025-08-29 by vvbandeira