The-OpenROAD-Project/OpenROAD-flow-scripts #3498: CMake Error while Building

cc-PLC, 2025-09-17:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Errors occur when running `./build_openroad.sh --local`

### Expected Behavior

Expect there are no error (see the log).

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.6.87.2-microsoft-standard-WSL2
os: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-24458-g96512a37e9
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 700 lines cut ...]
  src/par/CMakeLists.txt:18 (find_package)

-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### Additional Context

At the end of running ./setup.sh, it prompted to remove write-protected regular file:
```
Installed binaries to /home/leo/TSMC/JDP/OpenROAD-flow-scripts/dependencies/bin/verilator
Installed man to /home/leo/TSMC/JDP/OpenROAD-flow-scripts/dependencies/share/man/man1
Installed examples to /home/leo/TSMC/JDP/OpenROAD-flow-scripts/dependencies/share/verilator/examples

For documentation see 'man verilator' or 'verilator --help'
For forums and to report bugs see https://verilator.org

/tmp/DependencyInstaller-orfs-PrInbt ~/TSMC/JDP/OpenROAD-flow-scripts
rm: remove write-protected regular file 'verilator/.git/objects/pack/pack-7402ab73141cbe0d6d4529a9bc644e307b52d794.pack'? y
rm: remove write-protected regular file 'verilator/.git/objects/pack/pack-7402ab73141cbe0d6d4529a9bc644e307b52d794.idx'? y
~/TSMC/JDP/OpenROAD-flow-scripts
```
Is this normal?

maliberty, 2025-09-17:
Have you run the dependency installer?

maliberty, 2025-09-17:
I still have the same question - have y'all run the dependency installer and have abslVersion=20250814.0 installed?

@hongted @QuantamHD any ideas?

vvbandeira, 2025-09-17:
@maliberty I was able to reproduce and will investigate.

vvbandeira, 2025-09-17:
@cc-PLC @gadfort 
I think the issue is incompatible versions of `or-tools` and `abseil`.
Downgrading `abseil` to `20240722.0` fixed the issue locally for me.
I will make a PR to upgrade `or-tools` and `abseil` to a recent enough and compatible version, but in the meantime, you can replace it on the OR DependencyInstaller to get unblocked.
```
    abslVersion=20240722.0
    abslChecksum="740fb8f35ebdf82740c294bde408b9c0"
```

hongted, 2025-09-17:
From further investigation it doesn't seem to be an version incompatibility, but rather the

<deps_dir>/external/lib/cmake/absl/abslTargets.cmake is not defining all targets for the absel version in question.  Not sure why building ORtools from scratch triggers this and not when using a system version, but it may be that ORtools just uses more parts of absl.

hongted, 2025-09-18:
Root-cause of the `set_property could not find TARGET absl::*` errors is due to the following

1. Openroad's DependencyInstaller installs Abseil into the external dependencies dir https://github.com/The-OpenROAD-Project/OpenROAD/blob/e48606ad9a1220c831c2ef49e5d51b35e8fe1d2e/etc/DependencyInstaller.sh#L268 

2. If or-tools is not using a "system" version, then DependencyInstaller extracts an or-tools release into the external dependencies dir.  https://github.com/The-OpenROAD-Project/OpenROAD/blob/e48606ad9a1220c831c2ef49e5d51b35e8fe1d2e/etc/DependencyInstaller.sh#L369

3. It so happens that OpenRoad also depends on Abseil and the or-tools releases include their own Abseil version and cmake files.  Hence #2 will partially overwrite the installation of #1.  If #1 and #2 are the same Abseil version then there is no issue.  If they are of different versions then there may be a install partially of one version and that of another.  This is what is causing the issue. 

4. In the case of using a "system" version of or-tools there is no issue as even though there are two copies each are internally consistent so all targets (especially internal-only targets) are ok.

Solutions are following
1. The Abseil version in DepdendencyInstaller should track the Abseil version used in or-tools.

The current version used in OpenRoad is 9.11.4210 and depends on Abseil version 20240722.0 
https://github.com/The-OpenROAD-Project/OpenROAD/blob/e48606ad9a1220c831c2ef49e5d51b35e8fe1d2e/etc/DependencyInstaller.sh#L325C5-L325C22
https://github.com/google/or-tools/blob/ed94162b910fa58896db99191378d3b71a5313af/cmake/dependencies/CMakeLists.txt#L87

If we upgrade to or-tools version 9.13 we should update abseil to 20250512.0

2. Update the build to rely on the Abseil included in OrTools.  This is easily done for non-system installs of OpenRoad where we know where Abseil is going to be placed, but a "system" install of OpenRoad may not put Abseil in a discoverable place.

My thoughts are to go with #1 for now as suggested in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/3498#issuecomment-3303542207

mguthaus, 2025-09-22:
Doh, I just spent most of today on this issue. :/