The-OpenROAD-Project/OpenROAD-flow-scripts #3531: error with PDN: 0185

y20n20, 2025-09-30:
### Subject

[Stage]: Floorplaning.

### Describe the bug

Hello,
I am facing ERROR PDN-0185:
[ERROR PDN-0185] Insufficient width (3.78 um) to add straps on layer TopMetal1 in grid "CORE_macro_grid_1 - i_dpram_wrap_b0.i_RM_IHPSG13_1P_512x32_c2_bm_bist_10" with total strap width 6.6 um and offset 1.8 um.
Error: pdn.tcl, 5 PDN-0185

the problem that no other values change the "Insufficient width (3.78 um)" - so i guess i do not understand the  message..

### Expected Behavior

? working PDN

[pdn_fb_mem_top_ihp-sg13g2_base_2025-09-30_15-05.tar.gz](https://github.com/user-attachments/files/22622476/pdn_fb_mem_top_ihp-sg13g2_base_2025-09-30_15-05.tar.gz)

### Environment

```shell
attached is the pdn_issue, the config.mk and pdn_cfg.tcl
```

### To Reproduce

see above

### Relevant log output

```shell

```

y20n20, 2025-10-03:
Hi,
1. You can see the floorplan in the attachment.
2. its 5 macro spacing.
3. there is not ring .. is it mandatory ?
4. inspecting the sram-macros, they have no shapes to connect them - is this a problem ?
5. if needed I can send the complete issue
Thanks
[fb_mem_top_fp.pdf](https://github.com/user-attachments/files/22678335/fb_mem_top_fp.pdf)

y20n20, 2025-10-06:
I think i see the problem..
ever thou i place the macros at specific locations:

config.mk:
export DIE_AREA					=  0  0  1200  690
export CORE_AREA				= 20 20  1180  670
export MACRO_PLACE_HALO			= 5 5

sram_macro_placement.cfg:
i_dpram_wrap_b0.i_RM_IHPSG13_1P_512x32_c2_bm_bist R0 100 100
i_dpram_wrap_b1.i_RM_IHPSG13_1P_512x32_c2_bm_bist R0 650 100
i_dpram_wrap_b2.i_RM_IHPSG13_1P_512x32_c2_bm_bist R0 100 400
i_dpram_wrap_b3.i_RM_IHPSG13_1P_512x32_c2_bm_bist R0 650 400

export MACRO_PLACEMENT			= $(DESIGN_HOME)/$(PLATFORM)/${DESIGN_NAME}/sram_macro_placement.cfg

the end result is as in the png file.

[image]

my question: what can cause that ??

y20n20, 2025-10-06:
fixed the placement error.. see attached png 

[image]

but still getting [ERROR PDN-0185] :

define_pdn_grid grid
add_pdn_stripe grid/Metal1
add_pdn_stripe grid/Metal5
add_pdn_connect Metal1 <-> Metal5
define_pdn_grid CORE_macro_grid_1
add_pdn_stripe CORE_macro_grid_1/TopMetal1
[ERROR PDN-0185] Insufficient width (3.78 um) to add straps on layer TopMetal1 in grid "CORE_macro_grid_1 - i_dpram_wrap_b0.i_RM_IHPSG13_1P_512x32_c2_bm_bist_10" with total strap width 6.6 um and offset 10.0 um.
Error: pdn.tcl, 5 PDN-0185
Command exited with non-zero status 1
Elapsed time: 0:00.33[h:]min:sec. CPU time: user 0.31 sys 0.02 (100%). Peak memory: 99308KB.
make[1]: *** [Makefile:409: do-2_4_floorplan_pdn] Error 1
make: *** [Makefile:409: results/ihp-sg13g2/fb_mem_top/base/2_4_floorplan_pdn.odb] Error 2

maliberty, 2025-10-06:
Please upload the corrected test case

y20n20, 2025-10-06:
attached the relevant files.
Thanks

[pdn_fb_mem_top_ihp-sg13g2_base_2025-10-06_15-53.tar.gz](https://github.com/user-attachments/files/22726064/pdn_fb_mem_top_ihp-sg13g2_base_2025-10-06_15-53.tar.gz)

[sram_macro_placement.tcl.txt](https://github.com/user-attachments/files/22726085/sram_macro_placement.tcl.txt)

[configfiles.zip](https://github.com/user-attachments/files/22726099/configfiles.zip)

maliberty, 2025-10-06:
Please do `make pdn_issue` as that will package a complete test case and upload the result.

y20n20, 2025-10-07:
yes. [pdn_fb_mem_top_ihp-sg13g2_base_2025-10-06_15-53.tar.gz](https://github.com/user-attachments/files/22726064/pdn_fb_mem_top_ihp-sg13g2_base_2025-10-06_15-53.tar.gz) is the "make pdn_issue"

y20n20, 2025-10-08:
after reinstalling OR the error vanished.
thank you. you may close this item.